prusa3d/PrusaSlicer #8748: Request buildplate xyz Indicator visible during zoomed view while manually placing SLA support points

Subtle3D, 2022-08-23:
It is difficult to manually place SLA support points in ideal positions on an object without being able to tell which direction the build plate is.

A global xyz indicator always displayed somewhere, such as the top left or right corner of the display, regardless of camera position or zoom level would be ideal. The current indicator is locked to the build plate corner and is often out of view. 

I constantly have to zoom in and out to see what angle I'm looking at.
The attached picture is zoomed in for precise support point placement but I'm not sure which side is toward the build plate until zoomed out enough that point placement is difficult. 

[image]
[image]

kubispe1, 2022-08-24:
Hm hm, floating xyz compass / indicator could be sometime useful. Thanks

n8bot, 2022-08-24:
A viewcube would be nice. Implemented in dear imgui here, as an example: https://github.com/CedricGuillemet/ImGuizmo

closed 2026-09-29 by prusaslicer-bot[bot]