prusa3d/PrusaSlicer #8751: Stl doesn´t work

pfruiz, 2022-08-24:
### Description of the bug

3d model looks fine in FreeCAD.
stl looks fine in PrusaSlicer 2.4.2 after importing.
After "Slice now", part of text vanishes.

The vanished text was added to a step file using FreeCAD.
[cap.zip](https://github.com/prusa3d/PrusaSlicer/files/9414660/cap.zip)

### Project file & How to reproduce

3d model looks fine in FreeCAD.
stl looks fine in PrusaSlicer 2.4.2 after importing.
After "Slice now", part of text vanishes.

The vanished text was added to a step file using FreeCAD.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa mk3s MMU2S

kubispe1, 2022-08-24:
In 2.4.2 you can use "Thin wall detection", but the improvement is quite poor. You can get a much better result, if you use the newer release of PS - 2.5.0-rc1 with Arachne engine. Even though, side text is quite tiny and narrow, so some detail could be suppressed.

stepikovo, 2022-08-29:
This is not really an issue with slicing, the features on the model are simply too small to be printed with your default (0.4 mm) nozzle.

closed 2022-08-29 by stepikovo