prusa3d/PrusaSlicer #8752: Perimeters dissapear after slicer in view.

rsenas, 2022-08-24:
### Description of the bug

Perimeters dissapear after slicer in view.

### Project file & How to reproduce

[Perimeter.zip](https://github.com/prusa3d/PrusaSlicer/files/9416797/Perimeter.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-beta1+win64

### Operating system

Windows 10

### Printer model

Ender 5 Plus

rtyr, 2022-08-24:
I can see that perimeter visualisation is disabled (your screenshot).

Just click on the perimeter in the legend, it toggles visibility in the preview.

[image]

Answered, closing.

closed 2022-08-24 by rtyr

rsenas, 2022-10-11:
Great, thanks a lot.