prusa3d/PrusaSlicer #8753: Crash on on Slicing with Arachne

LukeL99, 2022-08-24:
### Description of the bug

When I slice this model with 2.5.0-rc1 on Windows 10 64-bit, Prusa Slicer crashes and silently exits. It does not crash if I switch from Arachne to Classic.

[image]

### Project file & How to reproduce

[Main Color Plate 9 - ASA.zip](https://github.com/prusa3d/PrusaSlicer/files/9418390/Main.Color.Plate.9.-.ASA.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-rc1+win64

### Operating system

Windows 10

### Printer model

i3 MK3S+ (Genuine)

kubispe1, 2022-08-25:
Hello, I noticed you have skirt out of bed :D , but it is not the point of the issue. Seems to me that there could really some trouble with Arachne. 
- This 3mf works properly in PS 2.5.0-alpha3.  
- In this specific example crash is caused by parameter - External perimeters width (Printer settings / Advanced / Extrusion width / External perimeters). For 0,4 mm it crash, For 0,45 do not crash.

- Could be similar issue as: #8724

hejllukas, 2022-08-30:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/de48bcd1b2b8c479f2c2000258a9cf72c5f48829 in master_250, and it will be part of the next release.

closed 2022-08-30 by hejllukas