prusa3d/PrusaSlicer #8756: Crash when selecting variable layer height

rrroonn, 2022-08-25:
### Description of the bug

The program crashes when the variable layer height icon is clicked.

### Project file & How to reproduce

Reproduce:
- Add a shape to the platter.
- Click variable layer icon.
Program crashes.

[Shape-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9421324/Shape-Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0-beta1+linux-x64-GTK3

### Operating system

Linux Mint 20.3 Cinnamon / Kernel 5.4.0-124-generic

### Printer model

Generic - using Marlin Legacy

kubispe1, 2022-08-25:
Hello, thanks for report. We test it on Win 10 and Linux Ubuntu and it works. Can I ask you to try it also in PrusaSlicer 2.5.0-rc1 ?

rrroonn, 2022-08-25:
Just downloaded  Version 2.5.0-rc1+linux-x64-GTK3

Same issue exists.

Maybe this might help?

`
slic3r_main[77764]: segfault at 0 ip 00007fdd28ea5a7f sp 00007fffd5e62d10 error 6 in i965_dri.so[7fdd289de000+a23000]

...

Code: ff ff ff 0f 1f 84 00 00 00 00 00 48 8b 46 10 48 89 d7 48 89 42 10 e8 c0 f9 ff ff 48 8b 45 08 48 89 2b 48 89 43 08 48 8b 45 08 <48> 89 18 48 89 5d 08 e9 3a ff ff ff 0f 1f 44 00 00 41 54 55 53 48

...

Process 77764 (slic3r_main) of user 1000 dumped core.

Stack trace of thread 77764:
#0  0x00007fdd28ea5a7f n/a (i965_dri.so + 0x62ca7f)
#1  0x00007fdd28ec0a31 n/a (i965_dri.so + 0x647a31)
#2  0x00007fdd293bdfb7 n/a (i965_dri.so + 0xb44fb7)
#3  0x00007fdd293c3515 n/a (i965_dri.so + 0xb4a515)
#4  0x00007fdd293c577c n/a (i965_dri.so + 0xb4c77c)
#5  0x00007fdd290fb081 n/a (i965_dri.so + 0x882081)
#6  0x00007fdd290fe70d n/a (i965_dri.so + 0x88570d)
#7  0x00007fdd2909d667 n/a (i965_dri.so + 0x824667)
#8  0x00007fdd28a39456 n/a (i965_dri.so + 0x1c0456)
#9  0x00007fdd28a39ea1 n/a (i965_dri.so + 0x1c0ea1)
#10 0x00007fdd28a352c9 n/a (i965_dri.so + 0x1bc2c9)
#11 0x00007fdd28a26de8 n/a (i965_dri.so + 0x1adde8)
#12 0x00007fdd28a27d76 n/a (i965_dri.so + 0x1aed76)
#13 0x00007fdd28bc3ddf n/a (i965_dri.so + 0x34addf)
#14 0x00007fdd28bc2dbb n/a (i965_dri.so + 0x349dbb)
#15 0x00007fdd28bc5024 n/a (i965_dri.so + 0x34c024)
#16 0x00000000010285ae n/a (/tmp/.mount_PrusaSKS5pmk/usr/bin/bin/prusa-slicer + 0xc285ae)
`
I have a core dump file if that helps.

Ron

rrroonn, 2022-08-25:
Just for completeness, I also tried `PrusaSlicer-2.5.0-rc1+linux-x64-GTK2-202208191500.AppImage` - same issue.

YuSanka, 2022-08-25:
Do you know last PS version, which works for you?

rrroonn, 2022-08-25:
Sorry, no, I have not printed in the past year or more and only recently
started printing again.  Last time I printed I was using Windows 10 but
have since migrated to Linux Mint.

Godrak, 2022-08-26:
Hi. We tested it with the same software setup - Linux Mint, same kernel version and gtk libs versions, but we could not reproduce it. 
However, from your log it is clear that the crash originates in i965_dri driver, and there is a known issue with some intel mesa drivers and variable layer height: https://github.com/prusa3d/PrusaSlicer/issues/7274 .
This problem will be solved in 2.6 release, in the meantime please use one of the custom builds mentioned in the issue, or you can build & try the current master.

rrroonn, 2022-08-26:
Thank you. Glad its being sorted out.

kubispe1, 2022-12-13:
In issue #7274 community has tested on experimental branches this issue. OpenGL 3.3 looks stable and it is fixed in 2.6.0.
So, I close this issue. Thank you

closed 2022-12-13 by kubispe1