prusa3d/PrusaSlicer #8760: Crash on slicing stl but slicing repaired stl is working

tcjtomas, 2022-08-25:
### Description of the bug

When loading a simple stl and slicing it fails.

Repair stl, load the saved obj file and slicing it works.

### Project file & How to reproduce

Load stl and slice causes silent crash

Load stl repair it

Save obj file

Slice is successful
[Adapter.zip](https://github.com/prusa3d/PrusaSlicer/files/9425336/Adapter.zip)
l

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0-rc1+linux-x64-GTK3

### Operating system

Operating System: openSUSE Leap 15.3 KDE Plasma Version: 5.18.6 KDE Frameworks Version: 5.76.0 Qt Version: 5.12.7 Kernel Version: 5.3.18-150300.59.90-default OS Type: 64-bit Processors: 4 × Intel® Core™ i7-7500U CPU @ 2.70GHz Memory: 7,7 GiB

### Printer model

Prusa i3 mk2 reprap

kubispe1, 2022-08-26:
Hello, thanks for you issue report. I think, your model is somehow broken (but just little bit).
But it doesn't apologize the crash of PS. I tried problematic .stl on WIN 10 and that worked. 
[image]
So, please can I ask you to try this steps for us?:
1) Try it in 2.5.0-rc1 with Classic perimeter generator (Print settings/ Layers and perimeters / Perimeter generator)
2) Also try it in 2.5.0 alpha3 with Arachne engine.
This would help us to detect or filter some cases.

hejllukas, 2022-08-30:
Hi, we fixed the artifacts shown on this model when Arachne was used with the commit https://github.com/prusa3d/PrusaSlicer/commit/614cd7081d4a33104272eb64d8f50939844ecbbd in master_250.
But we weren't able to reproduce the crash that you mentioned. Could you please try this attached build and report back if it is still crashing or not?
[PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.AppImage](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lh/PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.AppImage)
[PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.tar.bz2](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lh/PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.tar.bz2)
If it will still crash with this build, could you please upload 3MF exported before the crash?

tcjtomas, 2022-08-30:
Hi Team, 

I'll try later to day.

---
MVH/BR 
TOMAS JENSEN 
+4524278454 
WEB: FOTOSITE [6] 
WEB: VEJRET [7] 

Lukáš Hejl skrev den 2022-08-30 14:30:

> Hi, we fixed the artifacts shown on this model when Arachne was used with the commit 614cd70 [1] in master_250.
> But we weren't able to reproduce the crash that you mentioned. Could you please try this attached build and report back if it is still crashing or not?
> PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.AppImage [2]
> PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.tar.bz2 [3]
> If it will still crash with this build, could you please upload 3MF exported before the crash? 
 

Links:
------
[1]
https://github.com/prusa3d/PrusaSlicer/commit/614cd7081d4a33104272eb64d8f50939844ecbbd
[2]
***@***.***/files/lh/PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.AppImage
[3]
***@***.***/files/lh/PrusaSlicer-2.5.0-rc1+17-linux-x64-GTK3-g614cd70-202208301031.tar.bz2
[4]
https://github.com/prusa3d/PrusaSlicer/issues/8760#issuecomment-1231601369
[5]
https://github.com/
[6] https://tomas.dk/ps
[7] https://tomas.dk/Vejr

tcjtomas, 2022-08-30:
Hi Team,

It slices with PrusaSlicer-2.5.0-rc1+17. No issues. And futher more the artifacts are not showing.

Well done :)
This issue can be closed.

BR
Tomas Jensen

hejllukas, 2022-08-30:
Thank you for testing it and reporting back. So I'm closing this issue.