prusa3d/PrusaSlicer #8761: crashing when turning a large object on the printplate

dan-and, 2022-08-25:
### Description of the bug

PrusaSlicer 2.5.0-rc1 crashes when I try to turn a large object ( https://www.printables.com/model/262124-gridfinity-grid-for-kallax/files : Kallax_grid.stl ) on a slightly modified CR-10S Pro V2 printplate. 

Configuration of the Printbed ( especially: bed_shape = 0x0,308x0,308x305,0x305 ) 
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/9426754/config.zip)

Short video of the 90 degree turning and centering. 
[image]

Project file:
[Kallax_Grid.zip](https://github.com/prusa3d/PrusaSlicer/files/9426778/Kallax_Grid.zip)

### Project file & How to reproduce

1. import Kallax_Grid.stl from https://www.printables.com/model/262124-gridfinity-grid-for-kallax/files 
2. turn by 90° 
3. may be center and turn by 90° again 

prusaslicer crashes :-( 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc1

### Operating system

Windows 11

### Printer model

Sunlu S8 (modified CR-10 S Pro V2)

dan-and, 2022-08-26:
When I completely remove the skirt (set with a distance of 7 mm), it doesn't crash (and is really printable), so maybe it is caused by the calculation of the skirt at prints which are nearly exact size of the print bed

hejllukas, 2022-08-30:
Thank you for reporting this issue. Based on a report from testers, this was also fixed with https://github.com/prusa3d/PrusaSlicer/commit/de48bcd1b2b8c479f2c2000258a9cf72c5f48829 in master_250.

closed 2022-08-30 by hejllukas

dan-and, 2022-08-30:
I can confirm that it is fixed with -RC2 

Thanks for the quick support :-)