prusa3d/PrusaSlicer #8763: Infill Lighting speed

RJSavell, 2022-08-25:
### Description of the bug

Perhaps I am missing something but for a print when I choose rectilinear the estimated print time is 4:23 but set the infill to use Lighting the estimated time is 7:11. Shouldn't one of the benefits of using lightning be that it prints faster?

### Project file & How to reproduce

Select any the rectilinear and slice and note the time
select Lighting and slice and note the time
[Shamrock.zip](https://github.com/prusa3d/PrusaSlicer/files/9428506/Shamrock.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 RC1

### Operating system

Windows 11

### Printer model

Ender 5+

rtyr, 2022-08-25:
Expected behavior. Lightning infill produces much more retractions in this case. Retractions in your project file are also set to very slow speeds (3mm at 10/7 mm/s).  To improve the situation you can either change your retraction settings or enable "only retract when crossing perimeters" option (so PS will not perform retractions within infill). It may be still beneficial to use standard infill for this particular object.

Can we close the issue?

RJSavell, 2022-08-26:
Ok thanks and yes can close this.

closed 2022-08-26 by rtyr