prusa3d/PrusaSlicer #8764: Filament preset change on color change

Vojtak42, 2022-08-26:
Only color change is nice feature, but filaments have different temps etc. so it would be nice to have filament preset change option.

kubispe1, 2022-08-26:
Hi, I am so sorry, but I am not able to understand of your request. Can you please describe it more ? Also, please can you add purpose and maybe some example ?.... Change of materials is available at the moment (one example - printer profile Original Prusa i3 MK3S & MK3S+ MMU2S)

Vojtak42, 2022-08-26:
Yes, change of filament is avaible but there is no drop down menu in which I would be able to select to which color (preset) I want to change, because different filaments needs different temperature, have different density and price.

rtyr, 2022-08-26:
I believe this issue is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2002. I am closing this issue.

closed 2022-08-26 by rtyr