prusa3d/PrusaSlicer #8765: Parameters related to nozzle diameter

laptop-laptop, 2022-08-26:
Hello, I am using a 0.4mm nozzle on an i3MK3S+. Over the years, I have tuned many parameters to fit my project and my specific printer. If I change the nozzle to one with a different diameter, what parameters do I need to change manually? Thanks

kubispe1, 2022-08-29:
Hello, for your Prusa Printer i3 MK3s is a bunch of default printer profiles for other nozzle diameters in Configuration Wizard (go to Configuration/Configuration Wizard) https://help.prusa3d.com/article/configuration-wizard_1754.
It could be good starting point to tune other nozzles. But of course, for user needs you have to set it by your hand.
Also useful feature is Compare Presets ( Window/Compare Presets)
[image]

stepikovo, 2022-09-05:
You should be able to find all the answers here - https://help.prusa3d.com/article/creating-profiles-for-different-nozzles_127540

closed 2022-09-05 by stepikovo