prusa3d/PrusaSlicer #8768: SegmentationFault if window made too narrow

sdfgeoff, 2022-08-26:
### Description of the bug

If you resize the prusaslicer window to make it really narrow, then prusaslicer will crash with a segfault.

### Project file & How to reproduce

Resize it!
https://user-images.githubusercontent.com/13490050/186985383-31ff1569-b0cb-4d3d-a79c-09af764b1741.mp4

It's worth noting that the reproducability of this bug will depend on your window manager. GTK provides the ability to ask the window manager not to be resized below a certain amount (and prusa-slicer does this: in floating mode the window cannot be be shrunk enough to experience the bug), but this is a size request and some window managers (eg i3wm) allow shrinking below this requested value.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+UNKNOWN based on Slic3r (with GUI support)

### Operating system

ArchLinux

### Printer model

Snapmaker

DragonFinlay, 2022-09-02:
I'm not entirely sure whether this is actually an issue with PrusaSlicer itself or the fact that you can exceed shrinking limits using that window manager.
Did you find any other software that shows the same behavior?

sdfgeoff, 2022-09-02:
No other software that I can think of exhibits this behaviour. (Even other GTK programs are fine). Of course, at narrow sizes, many programs become awkward to use (eg buttons don't display all the text, lots of sideways scrollbars etc.) but they don't crash.

I use a 1920x1080 screen and if I try to have 3 things open next to each other (ie a window width of 640) then it crashes.

ngrigoriev, 2022-12-03:
I have just observed this behavior when starting 2.5.0 manually on Linux with i3wm. And then after reading this issue I have started it on an empty desktop and it did not crash. Indeed, it crashes when getting only 1/3 of 4K screen.

closed 2026-09-29 by prusaslicer-bot[bot]