prusa3d/PrusaSlicer #8770: Bridge infill inconsistently aligned with solid infill it supports

johnsonm, 2022-08-27:
### Description of the bug

I have a model with nine identical areas that require bridging, with solid infill on top of them. In the bridging layer, some but not all of the areas have the bridging running in the same direction as the solid infill above it.

I would expect the bridging and solid infill in adjacent layers to run perpendicular to each other. First, zoomed in to show the problem

Bridging layer:

[image]

Solid infill on next layer up:

[image]

Same two layers, zoomed out to show that it is inconsistent:

[image]
[image]

### Project file & How to reproduce

Upload failed; "Something went really wrong, and we can't process that file." 4.8MB should fit within the 10MB limit...

The 3mf is in the zip file at https://people.danlj.org/mkj/files/er11tray.zip and also directly available as https://people.danlj.org/mkj/files/er11tray.3mf

Load the project and slice it. Look at layers 13 and 14 (3.9mm and 4.2mm, respectively) to see the problem.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-rc1+linux-x64-GTK3

### Operating system

Fedora 36

### Printer model

SK-Tank

kubispe1, 2022-08-29:
Hi, thanks for your issue. 
Add 1. part - Inconsistent angle of bridging: I think it is cause by settings of bridging angle = 0 
[image]

Add 2. part – perpendicular orientation of bridge and solid layer: At the moment there is only possibility to correct it with bridge angle and fill angle. But you are probably expect automatic recognition with default settings of bridge angle = 0?…?

johnsonm, 2022-08-29:
Thank you!

I'd (naively) expect a bridging angle of the default 0 to be "I don't care but be consistent if there is no specific reason to bridge in another direction" and for the default direction of bridging to be opposite the dominant direction for layer orientation in the layer immediately above it. For setting an explicit bridging angle, it could easily be wildly wrong for some layers that need bridging; that you might reasonably end up with no ability to specify an explicit bridging angle that works through the model.

I assume that by default, a user would not want adjacent layers with the same orientation, except for overriding reasons, like substantially reduced gap size in one direction over another that is different for different features in a single layer.

I did not try to test this in older stable versions to know whether this is a change in behavior. I had never noticed it before, but haven't looked for it either. ☺

Anyway, none of this is meant to imply an obligation to do anything. Just trying to articulate my naive-to-this-code-base expectations.

kubispe1, 2022-08-30:
I agree with you. Thanks for your point of view. I don't feel it is bug, but some kind of improvement.

closed 2026-09-29 by prusaslicer-bot[bot]