prusa3d/PrusaSlicer #8771: API through which to download files from the WEB, crypted, directly to PrusaSlicer.

rainell, 2022-08-27:
Is it possible to develop an API for web applications, through which to download STL files from the WEB, crypted, directly to PrusaSlicer, inaccessible from explorer/finder or other applications? If possible, it would be necessary, strongly necessary, to have the possibility to communicate PrusaSlicer if the export and save functionalities should be enabled or disabled and that only the option to generate the gcode be enabled.

murk-sy, 2022-08-27:
I assume you are talking about DRM for models? To prevent piracy/sharing?

In that case, this is a duplicate of #8070.
See my reply on that issue for an explanation why it's a bad idea.

rainell, 2022-08-27:
It is not a duplicate, although it has the same function. Feature Request [#8070](https://github.com/prusa3d/PrusaSlicer/issues/8070) refers to a direct connection between Prusa website ([printables.com](https://www.printables.com/)) and PrusaSlicer. My request is about the possibility of doing the same from a WEB site owned by me or owned by others, through APIs (for example Javascript APIs).

rainell, 2022-08-27:
This Feature Request is not for big download sites like printables, thingiverse, cults3d, etc. is for artists, etc. who want to expose and sell their work exclusively through their own website and their own funnel.

murk-sy, 2022-08-27:
If I focus strictly on preventing model exporting to be resent:

When the model is loaded into the slicer, it gets decrypted. This means that even if you were to disable model export from the slicer itself, there is really no guarantee for a "hacked" slicer (I mean, it's open source, duh) to exist.

A way to avoid that is to have an online, cloud based slicer. Not really a very good solution, and likely to be unpopular with paying customers. Keep in mind that if you can see the model in 3D in your browser, you can already pirate it.

However, even if you make a cloud slicer, you would need to prevent that gcode from being leaked or used to reproduce models. You could very easily just make a very finely printed SLA model "gcode" and reproduce the model from that.

So basically, you also need a printer that is *also* locked down, so you can encrypt gcode from being misused or even used on another printer.

Oh, by the way - the pirates, the *not* paying customers? They wouldn't be limited by the slicer, or the printer. They would have total freedom to print, share, or back up those models beyond their source. This entire rigmarole would only affect the lowest skilled pirates and would need a massive time investment.

All this might not be problematic for every single user, but I think it goes against Prusa's philosophy, and against mine as well. Well philosophy, *and* wasting money on non-functional DRM, *AND* screwing over paying customers. There's plenty of parallels you can draw with video streaming services as well.

If you plan to sell models to the existing hobby 3D printing community (which is full of Enders and the like), you've got no choice but to use an honor system.

Once Printables integration is added eventually, you can expect more model sales just from the fact that it will be very convenient to use it.

As an addon:
You *can* fight pirates by fingerprinting models completely invisibly, by making small deviations on the model surface - this is something that is fairly trivial to automate. Once you find a pirated model, you can track down the user that pirated it by checking the fingerprint and ban them. Not perfect and quite easy to scrub by the pirate by just having 2 of the model, but it's a relatively easy step that doesn't affect your customers.

**tl;dr: Bad idea, strongly against it.**

rainell, 2022-08-27:
I understand and respect your point of view and your arguments. I appreciate the time you have spent on this request.

gudvinr, 2022-08-28:
No, please no DRM. This will just make life of average person worse because honest people are the ones who will take impact of DRM and not people who you target it with.  
People who steal models will just use fork of slicer with disabled DRM protection and you gain nothing from it.

bubnikv, 2022-08-29:
@rainell 

Such request does not make sense, because PrusaSlicer is an open source application, thus anybody could modify and recompile to gain access to your data.

DRM / signing / encryption is the only option. Signing / encryption is being discussed as an optional 3MF file format extension.

closed 2022-08-29 by bubnikv