prusa3d/PrusaSlicer #8772: [FEATURE REQUEST] Anti Warping Tabs

Ic33s, 2022-08-27:
**Is your feature request related to a problem? Please describe.**
I want to have brim only on specific spots (e.g. corners). Using additional flat cylindric parts is not the same because it becomes part of the model I want to print and is no easier to remove brim.

**Describe the solution you'd like**
1. I want a part modifier, which generates only on specific spots (e.g. defined by me) a brim around the 3D model I want to print.
2. Alternative: I want to paint the parts of my 3D model I want a brim generated for.
3. Example: Cura Plugins "https://github.com/5axes/TabAntiWarping" and "https://github.com/5axes/TabPlus". which are quite good at doing this job - beside on fact: They are using support settings to generate the tab, which is at least irritating.

**Describe how it would work**
Part modifier:
1. "Add modifier" to my 3D model - e.g. a cylindric one.
2. Place the modifier part anywere at the 3D model wall and touching the print bed.
3. Enable to modify brim settings (important first one would be: "Enable Brim", I'm not sure if further brim settings need to be changeable from the print settings profiles one)

Alternative -> Paint brim:
Like the part modifier, but instead printing the brim on my 3D modell. For the brim settings (e.g. width and seperation gap) are the default settings of the print settings profile used.

**Describe alternatives you've considered**
I tryed adding flat cylinders and place them on the edges but this ones become part of my 3D model which I absolutly don't want.

Best

neophyl, 2022-08-27:
Duplicate issue.  See https://github.com/prusa3d/PrusaSlicer/issues/8283

kubispe1, 2022-08-29:
Thanks for feature request and also thanks to @neophyl  to link it. I am closing it as duplicate

closed 2022-08-29 by kubispe1