prusa3d/PrusaSlicer #8773: Custom G-code for each tool

Drzet, 2022-08-27:
Currently PS only allows single TOOL CHANGE custom gcode to be added.

I would like to be able to add custom g-code for each of the tools

My machine has 2 extruders (swivel) and the head is able to go beyond the back of the bed, in Y direction.
I would like to set-up 2 areas, for each extruder, for wiping between tool changes.
There is only space for this on each side of the bed, not in the middle, therefore each extruder can only reach it's side wipe area.

There might be other scenarios where individual tool custom gcode may be useful.
Such an option is present in Cura.

bubnikv, 2022-08-29:
I believe you could implement that with {if} {elsif} {else} {endif} macros.
https://help.prusa3d.com/article/macros_1775
https://help.prusa3d.com/article/list-of-placeholders_205643#placeholders-available-in-all-custom-g-codes

Drzet, 2022-08-29:
Thank you, I will try to use the macro.
However it would still be much easier for many users, to have this feature in the interface

closed 2026-09-29 by prusaslicer-bot[bot]