prusa3d/PrusaSlicer #8774: Bridging failure with 0% infill 2.5.0 rc-1

dtryba, 2022-08-27:
### Description of the bug

Object with an overhang that needs a bridge to get the top layers.
If the infill is non 0%, the bridge is correct. If the infill is 0%, the bridging it to short
[image]
[image]
.

### Project file & How to reproduce

[bridgebug.zip](https://github.com/prusa3d/PrusaSlicer/files/9438780/bridgebug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc1+linux-x64-GTK3

### Operating system

Debian 11PrusaSlicer-2.5.0-rc1+linux-x64-GTK3-202208191449.AppImage

### Printer model

Prusa Mini

dtryba, 2022-08-30:
Not only the bridge is foobar, solid layers above it til the top infill are also.
But the "Ensure vertical shell thickness" is what triggers the corruption. When enabled the bridge and layers above are correct.

kubispe1, 2022-09-02:
Hello, there is missing to set on - Ensure vertical shell thickness (Printer settings and perimeters/Layers and perimeters/Ensure vertical shell thickness. 
Than it works in 2.5.0 rc-2. So, I am closing this issue at the moment.

closed 2022-09-02 by kubispe1

plampix, 2022-09-02:
So by changing a totally unrelated setting, the slicer produces unprintable g-code? I don't think that's acceptable, especially since it used to work.

dtryba, 2022-09-02:
How is this not a bug?

- A bridge shouldn't depend on infill that isn't even there. Setting infill to 0.01% the bridge is correct, even when there is no infill in the gcode.

- A bridge should not depend on "Ensure vertical shell thickness", especially not when there is no difference in the the gcode whether the option is enabled of not.

This bug/behavior exists in rc-2 and in 2.4.x. It results in failed bridging in simple of objects, potentially in messed up top surfaces if there aren't enough layers to correct for the failed bridge.

kubispe1, 2022-09-06:
Nice discussion, I afraid you are right. Thanks for explanation. I let it closed because, it is duplicate issue #8571 at a same time.

dtryba, 2022-09-06:
It might be closely related to #8571. I agree on the duplicate.

dtryba, 2022-09-07:
Oh, missed that 2.5.0 was released!