prusa3d/PrusaSlicer #8775: Speed settings seems to be ignored

erlis, 2022-08-28:
### Description of the bug

I have speed settings for infill, perimeter, etc, but with a specific model (I can provide STL) my print is printing very slow everything, like if the speed setting was ignored completely 

### Project file & How to reproduce

Everything prints at around 15mm/s 

[image]

My speed settings: 

[image]

this is the STL I'm getting the issue
[rod-long.zip](https://github.com/prusa3d/PrusaSlicer/files/9439888/rod-long.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc1

### Operating system

Windows 10

### Printer model

voron 2.4 with stealthburner

neophyl, 2022-08-28:
Probably the speed over rides under filament when the layer times are low. Please attach your zipped project file if you need this confirming.

erlis, 2022-08-28:
[slow-speed.zip](https://github.com/prusa3d/PrusaSlicer/files/9440088/slow-speed.zip)
I've uploaded the project so you can see if I have any setting wrong there, I search and couldn't find anything wrong

neophyl, 2022-08-29:
It is the setting I mentioned.  Under Filament Settings>Cooling>cooling thresholds there are settings that cause the print to be slowed down if the layer time is short.  With the fast vorons then this setting catches a lot of people out.

The values you have in there are 20 seconds and 15mm/s.  So if the layer time is less than 20 seconds it slows it down to a minimum of 15mm/2.  Change the 15 to 30 and re-slice and you will see that the speed changes to 30 instead of 15 as it will be using the new minimum.  All these settings are expert level ones so you will need to be in expert mode to see them.

Best bet is to change the 'slow down if layer print time' value to something like 5s or even less.   That will let it use your configured speeds.  I normally have all my customised filaments set to 5-6 seconds precisely for this reason.

erlis, 2022-08-29:
Yes, that was the issue! Thanks for helping out! This issue can be closed I guess

kubispe1, 2022-08-29:
Ok, thanks for discussion. See you

closed 2022-08-29 by kubispe1