prusa3d/PrusaSlicer #8777: Width property of the purge line still shown in legend after it has been disabled

RandomDude4, 2022-08-28:
### Description of the bug

When "View" is set to "Width" (to better see what Arachne is doing) and "Feature types" does not include "Custom". The purge line is not shown on the build plate, however the very wide purge line still affects the color scale of the Width legend, making all lines about the same color.

It appears this is older issue has not been closed prematurely:
https://github.com/prusa3d/PrusaSlicer/issues/8479

See comments and links in that issue as they are quite new. 
Also see this link which shows the issue:
https://forum.prusa3d.com/forum/prusaslicer/how-to-disable-viewing-of-custom-g-code-messes-up-width-preview-with-arachne-in-2-5/

### Project file & How to reproduce

1. Add any shape
2. Slice it.
3. Select "View" -> "Width"
4. Purge line should be dark red and all lines in your shape should be blue
5. Deselect "Feature Type" -> "Custom"
6. Purge line should now be gone from build plate, but colors have not updated.

Verification of color scale:
8. Remove the all custom G-code related to the purge line from the Printer Settings
9. Re-slice and observe a much better color scale where the lines in your print now show proper colors according to their width

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 RC1

### Operating system

Windows 10

### Printer model

Prusa i3 MK3

kubispe1, 2022-08-30:
Thanks for issue. I think, it is reasonable to change legend for only visible features of model. Hope there will be time to improve it.

RandomDude4, 2022-08-30:
> Thanks for issue. I think, it is reasonable to change legend for only visible features of model. Hope there will be time to improve it.

For other properties, such as volumetric flow rate, the legend already works like this. So it seems this has already been implemented for some things but not others.

xiphmont, 2022-12-18:
+1; affecting me too, very very relevant when tuning arachne.

xiphmont, 2022-12-30:
Patch is about as simple as it gets; makes me think there's probably some good reason that I don't see why it isn't already like that.

kubispe1, 2023-05-17:
PS 2.6.0 beta1 
A new ‘Hide/Show Custom G-Code’ button was added to the G-Code preview. The range of color scale in the legend is possible to visualize without the effect of the purge line. The button is available only for parameters where it is relevant (Width and Volumetric flow rate)
[image]

Iam closing this issue

closed 2023-05-17 by kubispe1