prusa3d/PrusaSlicer #8779: Merged parts still printing independently

troypouliot, 2022-08-29:
### Description of the bug

I have a customized model that requires another standard part to be added to it. The 2 parts overlap and I need them to be merged together rather than printed each completely.

I can't share a complete view as this is a customer order.

This is the customized part;
[image]

This is the standard part added to it via right clicking the customized model --> Add Part --> Load...
[image]

When the model is sliced, I end up with this where the 2 models print over one another;
[image]

My current workaround is to export the model as a stl, then import this stl and slice.
[image]

[image]

### Project file & How to reproduce

Cropped down project file attached.

1. open project
2. slice

[merge models issue.zip](https://github.com/prusa3d/PrusaSlicer/files/9446013/merge.models.issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-rc1+win64

### Operating system

Windows 10

### Printer model

voron 0

kubispe1, 2022-08-30:
Hello @troypouliot, I think that you disabled Clip multi-part objects (Printing settings/Advanced/ Other - Clip multi-part objects) Than your merged multipart object is sliced like single object.

troypouliot, 2022-08-31:
@kubispe1 , Thank you! You are correct, I had that set wrong.

closed 2022-08-31 by troypouliot