prusa3d/PrusaSlicer #8781: Lightning infill not correctly using modifier fill density

edwardhorsford, 2022-08-29:
### Description of the bug

I have a model with two separate modifiers, each applying lightning infill. You can't seem to set the infill density differently between the two modifiers. The density value seems to get shared between the modifiers - whichever is denser.

I discovered this whilst making some infill samples - my samples generate infill at 10, 15, and 20% - but for infill they all look the same - as you can't seem to vary the density per modifier.

### Project file & How to reproduce

Example case, with simple box and two modifiers:
[image]

With both modifiers set to lightning infill and 5% density:
[image]

With the right modifier set to 50% density and the left still at 5%:
[image]

[Lightning-infill-bug.3mf 2.zip](https://github.com/prusa3d/PrusaSlicer/files/9448132/Lightning-infill-bug.3mf.2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-beta1

### Operating system

MacOS 11.6.8

### Printer model

Mini+

hejllukas, 2022-08-30:
Hi, this behavior is intentional. The reason is that way how Lightning infill is designed, it needs to be computed over the entire model with the same infill density for the entire model.
But with more than one infill density modifier (the object itself also counts as one modifier), we need to somehow deal with having more than one infill density for the model. We decided for those cases to use the arithmetic average of all densities of all modifiers with Lightning infill. 
So in the case one modifier uses the Lightning infill and density of 50%, and the second one uses Lightning infill and density of 5%, then the resulting density for Lightning infill will be 27.5% for all modifiers with Lightning infill.

Lightning infill isn't the only infill that behaves like this. Adaptive Cubic and Support Cubic behave the same way for exactly the same reasons.

edwardhorsford, 2022-08-30:
@hejllukas thanks for the explanation! I wonder if it's worth an alert or something / or a message somewhere to note that infill density is shared like this. In my case it doesn't really effect me as I was only making swatches.

Will close this issue as not a bug.

closed 2022-08-30 by edwardhorsford