prusa3d/PrusaSlicer #8785: Route travel inside part, like in S3D

3DCoded, 2022-08-30:
**Is your feature request related to a problem? Please describe.**
I've heard of the feature to route travel inside part in S3D, but I use PrusaSlicer. Is there an equivalent setting to this?

**Describe the solution you'd like**
Implement "Route travel inside part" option

**Describe how it would work**
This would be a checkbox, maybe on a per-object basis, that would help reduce stringing.

**Describe alternatives you've considered**
I can't think of any except for buying S3D, but I feel at home in PrusaSlicer

**Additional context**
Maybe even be a slider like variable layer height, between speed, and less stringing.

jmd73, 2022-08-30:
Do you mean "Avoid crossing perimeters"?

It is under "Print Settings"/"Layers and perimeters"/"Quality"

[image]

stepikovo, 2022-08-31:
@3DCoded please let us know if this is what you are looking for so we can close it.

3DCoded, 2022-08-31:
Thank you! This fixes that.

closed 2022-08-31 by 3DCoded