prusa3d/PrusaSlicer #8786: "Place on face" does not work with small objects

darthgloda, 2022-08-30:
### Description of the bug

The "place on face" feature does not show the white clickable surface to place a given place on to the print bed if that object is very small (in my case, 2.6x2.8x2mm). Scaling the object by 120% or more solves the problem. Scaling the object by 110% is not enough to solve the problem.
In my case, I have split one STL into two objects. I have not tested the issue on a single-object file.

### Project file & How to reproduce

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/9454614/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0+x64 macOS and 2.5.0 rc2 macOS

### Operating system

macOS 12.4 (21F79)

### Printer model

Prusa Mini+

kubispe1, 2022-09-01:
Hi, faces are probably under detection threshold as you discovered. It is known problem, but thanks for report, we will keep it in mind.

DragonFinlay, 2022-09-07:
Similar to #6540

kubispe1, 2022-09-07:
I close it like a duplicate.

closed 2022-09-07 by kubispe1