prusa3d/PrusaSlicer #8788: Error in "wipe_tower_widht"

Alexandreasm, 2022-09-01:
### Description of the bug

Error in the "wipe_tower_widht" function when setting up, not accepting the entered parameters

### Project file & How to reproduce

[Color_Cali-Dragon_v1_Keychain.zip](https://github.com/prusa3d/PrusaSlicer/files/9465869/Color_Cali-Dragon_v1_Keychain.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusa 2.5.0 -rc2

### Operating system

Windows 11 64 bits

### Printer model

Kywood Tycoon IDEX

stepikovo, 2022-09-06:
Thanks for the report, it can be reproduced in 2.4.2 too.

Alexandreasm, 2022-09-07:
Hi,

Thanks for your response.  Yes, in version 2.4.2 the same problem occurs, I can't adjust the sizes coorectly in wipe_tower_widht.

Regards

Alexandre Magalhães

closed 2026-09-29 by prusaslicer-bot[bot]