prusa3d/PrusaSlicer #8789: Monotonic sequential print

Dakkaron, 2022-09-01:
**Is your feature request related to a problem? Please describe.**
Sequential print currently only allows prints that actually fit under the gantry. That means, I cannot print 2-3 tall objects with a small footprint sequentially.

**Describe the solution you'd like**
It's phyiscally possible to print multiple taller objects, by printing them back-to-front or vice versa. So e.g. first the one on the rear of the bed, then the one in the center and then the one in the front.

**Describe how it would work**
When Monotonic Sequential Print is activated, the slicer makes sure that the parts are printed in a monotonic direction (so if object A is in the rear, B in the center and C is in the front, it will print A B C or C B A but never A C B or C A B).

Since some printers might have an Y gantry (I've never seen one, but I guess it's possible), the user should be able to select in which axis the sequential prints should be ordered. 

The user also has to enter the clearence needed for the gantry in the direction of the axis (so the minimum distance between prints) and the preferred printing direction.

The preferred printing direction (back-to-front/front-to-back/left-to-right/right-to-left) is important, since, depending on the position of the nozzle on the print head, the minimum clearance needed might be smaller/larger. For example, if the nozzle is close to the front of the print head, but far from the rear, printing front-to-back will need much less clearance than back-to-front.

**Describe alternatives you've considered**
The advantage of monotonic vs regular sequential print is that monotonic sequential print can print taller objects sequentially.

The advantage of regular sequential print is, that you can arrange prints in both dimensions, thus fitting more objects on the printer.

So these two options could be both active in the same profile, and Prusa Slicer can automatically switch between them. The logic for switching would be the following:

- Does any part violate the clearances of regular sequential print?
  - no: use regular sequential print
  - yes: Does any part violate the monotonic sequential print gantry clearance?
    - no: use monotonic sequential print
    - yes: Display an error to the user, as it's currently done if the clearances of regular sequential print are violated.

The only workaround that currently exists is disabling the sequential print clearance checks and manually make sure the positioning and print order is right. This is error-prone and can lead to damage on the parts and the printer.

gudvinr, 2022-09-01:
> Since some printers might have an Y gantry (I've never seen one, but I guess it's possible), the user should be able to select in which axis the sequential prints should be ordered.

Some printers have cross-shaped X and Y rails moving above bed.

reify3d, 2023-02-19:
Please consider this request as delta printers such as the TRILAB DeltiQ 2 and AzteQ Industrial are unable to print sequentially.

closed 2026-09-29 by prusaslicer-bot[bot]