prusa3d/PrusaSlicer #8790: PrusaSlicer Is not working on my pc

Faizi57, 2022-09-01:
### Description of the bug

Everytime i change the layer quality and infill percentage, my software crashed.

### Project file & How to reproduce

[N YY ALLOY SALTPOT.zip](https://github.com/prusa3d/PrusaSlicer/files/9469360/N.YY.ALLOY.SALTPOT.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 11 pro

### Printer model

Prusa i3 MK3+

neophyl, 2022-09-01:
I just tried your project on the same version of PS but installed on windows 10 and it will slice and change those settings fine without crashing.
I should point out though that the model is broken and has 974 open edges.  This is resulting in what should be an open interior space being filled with infill for a large area of it.

It may be that the broken nature of the model is causing your computer the issue, so try repairing it first.  The built in repair via netfabb is not able to repair it though so it will probably require manual repair

Faizi57, 2022-09-01:
But I tried on an okay model
This happens again

kubispe1, 2022-09-01:
> Everytime i change the layer quality and infill percentage, my software crashed.

So, only for this changes Does it crash? Are you able to slice at all?

Faizi57, 2022-09-01:
Yaa i do slicing
But not changing infill percentage and layer height

neophyl, 2022-09-01:
How are you changing those 2 items ?  Are you changing the infill via the right hand panel or are you going in to the print profile and changing infill there ?  Same with layer height, are you choosing a complete new profile from the drop down or are you changing the value in the print profile ?