prusa3d/PrusaSlicer #8791: Printer setting - more extruders

micblazek, 2022-09-01:
### Description of the bug

I would like to use other extruder for supports and oteher for own object. 

- I select SULUBLE FULL printer setting 
- I try change extruder on  "Printer setting - more extruders"  
  - perimetrs, infil and full infil dont work, after slicing is still used extruder number 2
  - selection for support works fine 

### Project file & How to reproduce

You need MMU printer

Select 0.20mm SULUBLE FULL profill
Insert some model with spece for support
GO to page "Printer setting" - "More extruder"
Change parameter "Extruder for perimetr", "Extruder for infill" and "Extruder for full infill" 
Slice it

My settings and output is on screenshots

[image]
[image]
[image]
[image]

[Test_01_disky_3D.zip](https://github.com/prusa3d/PrusaSlicer/files/9471418/Test_01_disky_3D.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

Windows 10

### Printer model

MK3s+ MMU2s

rtyr, 2022-09-14:
I was able to reproduce it. I can see there is extruder 2 assigned in the model config for some reason.

[image]

rtyr, 2022-09-14:
@micblazek It should work fine if you remove the model and load it again from STL. For some reason, there is extruder 2 assigned inside your provided 3MF (it is not visible in UI). It could maybe happen because of some bug included in previous PS versions.

micblazek, 2022-09-14:
Thank you, I try it again and in PS 2.5 is working fine.

closed 2022-09-14 by micblazek