prusa3d/PrusaSlicer #8792: Windows Controlled folder access = Unhandled unknown exception; terminating the application

beepboop1234, 2022-09-01:
### Description of the bug

Saving any file causes the error "Unhandled unknown exception; terminating the application".

[image]

### Project file & How to reproduce

Upgraded from 2.5.0-rc1 to 2.5.0-rc2 win64. I have the Windows-included "Ransomware protection" enabled. If I don't "approve" the application it won't let the application make changes to most folders, including saving existing files after making changes. In every previous version of PrusaSlicer I would attempt to save, it would give an error it can't write to the location, and then I'd get a Windows pop-up telling me the application was blocked where I can then approve it. In rc2 that's not the case, it seems to crash and not give me a chance to approve it through Windows. Once I approved the application this stopped happening. Not project specific, happened on the three different projects I tested before approving it through Windows' "Ransomware protection". 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-rc2+win64

### Operating system

Windows 10

### Printer model

Prusa Mini+

bubnikv, 2022-09-02:
We will test the "Ransomware protection" Windows feature.

Frankly I don't think we changed anything between rc1 and rc2 regarding the ransomware protection behavior. Most likely your system was updated by Microsoft and it behaves differently now.

We have to reproduce first, then we should be able to take some action.

Our Windows development builds are not signed. We plan to update our build server to sign each development build produced, we hope that the Windows security features will be more benevolent then.

kubispe1, 2022-10-11:
Hi, we signed Win build already. Please can you try it if it will help for your issue..?
- https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0
- There you can find: PrusaSlicer-2.5.0+win64-202209060714_signed.zip

pniewiadowski, 2023-03-21:
I have faced that issue today with Version 2.5.1+win64 of PrusaSlicer.
How can I help to resolve that issue?

diegofreiiitas, 2024-10-23:
Did anyone manage to solve it?

pniewiadowski, 2024-10-23:
Just update the application

closed 2026-09-29 by prusaslicer-bot[bot]