prusa3d/PrusaSlicer #8793: Fuzzy skin on large cylindrical shapes using Arachne have a vertical 'seam line'

WDeenik, 2022-09-01:
### Description of the bug

When slicing bigger cylindrical objects using Arachne and adding fuzzy skin, the fuzzy skin generates a 'seam line' at a certain spot. It is not the actual seam of where the layer is started though, I explicitly painted the seam line in another place to make sure that wasn't the problem.

When I switch to the Classic perimeter generator, the problem disappears.

[image]

### Project file & How to reproduce

Slice the given project file with the given configuration. Rotate the object until you spot the seam line. Best visible from an angle from the top, see screenshot above. Seam line appears on the right side for me.

[Arachne_fuzzy_skin_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/9472958/Arachne_fuzzy_skin_bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc2+win64-202208301346

### Operating system

Windows 10

### Printer model

Prusa Mini

DragonFinlay, 2022-09-02:
Thanks for reporting!
I can reproduce this in both PrusaSlicer 2.5.0-RC2 as well as Cura 5.1.0, and it indeed doesn't happen with classic perimeter generation. So it seems to be related to Arachne.

hejllukas, 2022-09-02:
It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/8a69612fe1a6501b728c3027cbcff145f8fc789a in master_250, and it will be part of the next release.

closed 2022-09-02 by hejllukas