prusa3d/PrusaSlicer #8794: Maximum Y-speed for i3 style designs

3DCoded, 2022-09-01:
**Is your feature request related to a problem? Please describe.**
When printing tall models on i3 style printers, they will tend to wobble a little bet on the Y-axis.

**Describe the solution you'd like**
A speed setting to limit the Y-speed for tall models

**Describe how it would work**
It would be an Advanced feature in the Speed <- Print Settings menu.

**Describe alternatives you've considered**
I could manually edit the G-code, but I am not experienced with that.

**Additional context**
I am using a Prusa MK3S+ and PrusaSlicer 2.5.0-beta1

neophyl, 2022-09-02:
Why aren't you limiting that in the machine limits if its effecting you ?  Printer Settings>Machine Limits has values for Maximum Y feedrate and Maximum Acceleration Y.  There are also separate jerk values for Y too.

3DCoded, 2022-09-03:
Thank you! I was too afraid to touch those settings before, so now I'll create a profile for tall thin models.

closed 2022-09-03 by 3DCoded