prusa3d/PrusaSlicer #8797: Cannot build version_2.5.0-beta1 or higher (OCCCTWrapper errors) in Debug x64.

madmaxed, 2022-09-01:
### Description of the bug

Cannot build in debug VS Studio 2019 due to OCCCTWrapper errors
_ITERATOR_DEBUG_LEVEL '0' doesn't match value '2' in OCCTWrapper.obj

[image]
Release x64 compiles and runs fine.

Used both build_win.bat script and manually compiling dependencies.

Version 2.4.2 through alpha3 successfully running in Debug

### Project file & How to reproduce

[ALL_BUILD.zip](https://github.com/prusa3d/PrusaSlicer/files/9473844/ALL_BUILD.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

version_2.5.0-rc2

### Operating system

Windows 11 x64

### Printer model

Prusa MK3S+, SecKit Go2

bubnikv, 2022-09-02:
This issue will be fixed in PrusaSlicer 2.6.0.

madmaxed, 2023-02-22:
@kubispe1 As of Alpha3 it seems to be working.
Thanks.

closed 2023-02-22 by madmaxed