prusa3d/PrusaSlicer #8798: Permission issue with PrusaSlicer-2.5.0-rc2+linux-x64-GTK3-202208301345.AppImage"

RonRN18, 2022-09-02:
### Description of the bug

I have been using the 2.5.0 linux-x64-GTK AppImage since Alpha3 and have not had any issues launching the app in Ubuntu until downloading rc2. When I attempt to run the app, I get an error of "Cannot open /proc/self/exe: Permision denied" and "Failed to get fs offset for /proc/self/exe". I could probably start it if I ran it as a super-user (sudo) but I would rather not run such an app as a super-user. Again, this was not a problem in "Alpha 3", "Beta 1", or "RC1", I have only had this in "RC2."

### Project file & How to reproduce

From file directory: "./PrusaSlicer-2.5.0-rc2+linux+64-GTK3-202208301345.AppImage"

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-rc2+linux+64-GTK3-202208301345.AppImage

### Operating system

Ubuntu 22.04.1

### Printer model

Ender 3

dtryba, 2022-09-07:
An "strace -f ./Prusa..." might yield something interesting?

dtryba, 2022-09-07:
Oh, 2.5.0 is released.

lukasmatena, 2024-02-19:
Does not sound like PrusaSlicer issue. Closing.

closed 2024-02-19 by lukasmatena