prusa3d/PrusaSlicer #8804: Ironing mode

zsolt320i, 2022-09-04:
Dear Developers,

i'm love the ironing feature of prusa slicer, but in some cases the ironing method is not optimal.
For example if you have an "L" shape you would like to iron, the ironing on both leg is on the same direction, either in x or in y direction. That cause that on leg of the L shape is ironed easily and smooth along the long side of the leg (lengthwise), but the other leg ironing is difficult because the ironing is done across the width of the leg causing the 3d printer vibrating.

**Describe the solution you'd like**
I would implement the ironing always along the long/longer side, or following the shape contour.

**Describe how it would work**
For example the L shape ironing would be:
- in case of long side ironing - first it would be ironed the linf leg of the L along long side, after the short leg of the leg along long side
- in case of following the contour - the ironing would be done exactly as the printing was.

B.r.
Zsolt

neophyl, 2022-09-05:
If you change the direction of ironing you risk having different surface sheen when viewed from different angles.  As such I do not think this is a good idea.  Please keep it all the same direction.

zsolt320i, 2022-09-05:
> surface sheen

I don't think so, anyway a good compromise would be if you could choose from different method e.g: normal, long side, contour.
BTW, the "contour ironing" for sure would not cause different shine surface.

codefaux, 2022-09-06:
> > surface sheen
> 
> I don't think so, anyway a good compromise would be if you could choose from different method e.g: normal, long side, contour. BTW, the "contour ironing" for sure would not cause different shine surface.

Could not disagree more.

Movement direction is critical to surface finish, that's why Monotonic infill exists in the first place. It's a well established fact within the advanced 3D printering community, not the conjecture of a few commenters. You cannot change direction during printing solid surfaces, especially ironing, and expect a good finish.

"Vibration" during ironing shouldn't be an issue - that's why it defaults to going so slow, and moving in such tiny steps. Reasonable Acceleration settings should allow for any print at full speed not to disturb your machine, as well as reasonable Jerk settings -- that's why those options all exist.

Also, consider rotating your part 45 degrees. Then it won't need to change direction or spend 50/50 on good/"bad" operation, and it'll be more likely to look proper. Knowing which orientation to print an object is just as important as understanding your printer, its configuration, and all of the settings.

zsolt320i, 2022-09-07:
You don't have to be offensive!
For example, because "It's a well established fact within the advanced 3D printering community, not the conjecture of a few commenters" doesn't mean that the majority has right and the minority not.
I agree that it would be not easy to implement, but i do not agree that you know the result without trying (or even simulate) it.
If you are a developer and still do not see the sense than reject and close the topic.

closed 2026-09-29 by prusaslicer-bot[bot]