prusa3d/PrusaSlicer #8807: Bridging "quality" depends on number of perimeters

kozakm00, 2022-09-05:
### Description of the bug

See provided 3mf file.

Set Number of perimeters to 6 and observe bridging in layer 57 (layer height 0,2 mm). Notice overhang perimeters:
[image]

Set Number of perimeters to 3, bridging no looks ok, no overhangs:
[image]

Expectation: no mater what number of perimeters, bridging will always look correct, as on the second picture.

### Project file & How to reproduce

[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/9489161/bug.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-beta1+win64-202208151053

### Operating system

Windows 30

### Printer model

RR400K

stepikovo, 2022-10-19:
I'm closing this as a duplicate of #8034

closed 2022-10-19 by stepikovo

kozakm00, 2023-08-24:
Tested in Prusaslicer 2.5.0 and still valid. Reopening