prusa3d/PrusaSlicer #8809: Parameter modifiers also change part geometry during slicing

gudvinr, 2022-09-05:
### Description of the bug

When you apply settings using modifier volume and this volume happens to cross insides of the part, it adds perimeters inside and cuts part from external shell.

In example below I change speed of perimeters.  
Expected outcome would be that speed inside this modifier changes and that's it.
But as you can see in slicer preview, there is a part of this modifier inside object.

As far as I can say, it also happens with perimeter count and probably with other modifiers too.

[image]
[image]

### Project file & How to reproduce

1. Add any object
2. Use modifier to apply parameters for object part. Modifier should end inside object
3. Slice object

[xyzCalibration_cube.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9491564/xyzCalibration_cube.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Arch Linux, KDE Plasma

### Printer model

Artillery Genius Pro

neophyl, 2022-09-05:
Unfortunately that’s the way they work and have always worked as it’s how PRUSA implemented them. In effect it splits them into different volumes for slicing purposes. 
If you had done a search of closed issues you will find many similar ones on this topic already.

DragonFinlay, 2022-09-15:
Yes, this is a duplicate of #8121 and many other closed issues here. As neophyl said, this is how modifiers are currently handled.
It doesn't happen for all settings, but merely the ones changing parameters we can't yet influence during printing a single line.

For example, if you change the number of perimeters for a modifier, how are the perimeters supposed to split/merge? This requires a more advanced algorithm that we have yet to develop, test and implement.

closed 2022-09-15 by DragonFinlay