prusa3d/PrusaSlicer #8810: Progressive seam

espr14, 2022-09-05:
**Is your feature request related to a problem? Please describe.**
Sometimes there's no corner to place the seam. Seam causes visible dent or excess of filament.

**Describe the solution you'd like**
In case of circular external perimeter the beginning can start progressively and end progressively. This would make seam less visible. 

**Describe how it would work**
1. Print internal perimeters
2. Start external perimeter near the internal one and progressively go to proper distance from the internal one while increasing flow to the proper one (defined by the user).
3. End the perimeter with overlap with the beginning while progressively decreasing the flow.

**Describe alternatives you've considered**
You can progress in Z axis but this is more tricky because you would have to start with external perimeter first.

**Additional context**
The progressive end of the external perimeter would also be part of retraction (reduces stringing).
The image is just an approximation. The left part is more precise to the proposition.
[image]

kubispe1, 2022-09-06:
Interesting idea. But my opinion is that overlap transition will be even much more visible than standard solution...  
[image]

zsolt320i, 2022-09-06:
This method in some cases is applied in the industry. I see the point.

espr14, 2022-09-06:
> Interesting idea. But my opinion is that overlap transition will be even much more visible than standard solution...
PrusaSlicer does not allow thin lines at perimeter (so it's not in the image) but a 3D printer is able to extrude diameter that is much thinner than the diameter of a nozzle. Also this helps with progressive (de)pressurization of the nozzle.

gudvinr, 2022-09-06:
It looks like viable option with inclusion of Arachne engine. Is there some example gcode that can do that and pics of the result?
That could be doable by simply generating test object of some sort.

bubnikv, 2022-09-07:
@espr14 Dear Štěpán. That is a great idea. Please provide a pull request.

espr14, 2022-09-07:
@bubnikv Dear Vojtěch. Please merge my current pull requests first. Thank you.

bubnikv, 2022-09-07:
> @bubnikv Dear Vojtěch. Please merge my current pull requests first. Thank you.

Dear Štěpán.

Which one? There was no pull request provided for this another great idea of yours.

Warm regards,
Vojtěch Bubník

closed 2026-09-29 by prusaslicer-bot[bot]