prusa3d/PrusaSlicer #8818: Voids in layers when layer decreases in area

alidaf, 2022-09-07:
### Description of the bug

Slicing an object with a vertical extrusion that has a smaller area than its base leads to strange voids in the top layers of the base, even with 100% infill. The voids are where the areas of the extrusion are above some hidden ratio. These cause noticeable print artefacts associated with the old rectilinear top and bottom layers (before monotonic), which are especially noticeable when the voids are sharp.

The behaviour can be changed by adding the extrusion as a part to a blank base with a different infill %. In this case, the top layers (4) assume the outline shape of the added part. However, having the same level of infill in both parts leads to the voids reappearing.

Using modifiers with the original model to adjust the infill percentage of the base and extruded parts has no effect.

Object being sliced.
[image]

Top layer showing 3 voids that coincide with the larger areas of the extruded section.
[image]

Zoom of one of the voids
[image]

Top layer when adding the extrusion as a part with different infill %
[image]

### Project file & How to reproduce

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/9505477/Archive.zip)

Project files for original extruded model and with a base plus part.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+x64

### Operating system

macOS

### Printer model

MK3S

DragonFinlay, 2022-09-16:
This is intended behavior. The purpose of this is to connect different shapes through infill, connecting them much more effectively. If the top layers of the bottom object would be printed fully, the top part would only be connected through its circular base area and could easily be snapped off.

[image]

In your case the connecting areas are very small because this feature doesn't occur below a certain threshold / size of the top object.
You were mentioning that this causes "noticeable print artefacts" - I'm not entirely sure what you mean by that. The finished print should show no signs of these voids.

DragonFinlay, 2022-09-22:
I see what you mean now. I took another close look at your model (the one where base and text are separated) and the top infill lines of the base stop right at the perimeters of the text because the text is a few layers deep inside the base. Since these objects are not merged, PrusaSlicer treats them as separate objects and attempts to print both of them, even though they overlap.

If you move the text up a little bit so it sits on top of the object (this doesn't have to be extremely precise), the behavior matches the original part with only the small "voids" being present.

Regarding shear strength, let me give you an example of what I mean. Take this thumbscrew and check the transition of the threaded shaft to the head below:

[image]
[image]
[image]

The bottom solution will give you higher shear strength, it's quite difficult to break off the threaded shaft compared to the upper solution. So I don't agree that strength _always_ decreases with infill percentage.

Also, at very high percentages the material becomes more brittle. I'd never go above ~60% if there's no good reason for it.

alidaf, 2022-09-23:
Hi, Thanks for your input.

Your empirical observations don't match mine with threaded parts but it is a little disingenuous because I always print small threaded parts with contours only, rather than have any infill pattern. When I started printing these (presumably the nuts for the camera mount) I had more failures with infill than solid, and my threaded parts always failed at the start of the threaded region. I also found PETG to be far better than ABS.

From an engineering POV, shear stress is dependent on area - the larger the area, the lower the stress. The same with axial stress. The outer stress from bending is dependent on the 2nd moment of area and will still be higher for an annular cross section than a solid. Torsional shear stress is dependent on the polar moment of area (equivalent to 2nd moment of area for round sections) and, again will be higher for an annulus than a solid rod. 

Unfortunately, 3D printed parts are more orthotropic than isotropic (much weaker between layers) but the rules above still apply in general. Threaded parts will be under axial (z) and torsional load, both of which will result in lower stresses with the solid cross section according to traditional mechanics. The interaction between loads will create complex stress patterns, especially in orthotropic materials with a threaded geometry. 

BTW, solid Infill is layed down in orthogonal layers, which means it is stronger in the principle directions but weaker at points in-between. This is why I use contours only and could be why our observations don't match.

So the question remains, why in-filled parts for threads could be stronger than solid? I haven't run any finite element analysis on threaded parts with orthotropic materials and it would be difficult to model infill, but I can possibly accept that infill (within certain limits) increases the strength of threaded parts by redirecting load away from the weakest component (inter-layer), especially if using some of the 3-dimensional infill patterns. There are probably a number of factors that affect the strength but it obviously isn't as straightforward as traditional mechanics can determine when infill is involved.

But all of this is moot since my print is not load bearing. The only loads may be from residual stresses from non-unform cooling, but these will be negligible. I also disagree with your claim that the material becomes more brittle with percentage infill without some explanation. Heat, maybe. Print speed, maybe. Infill percentage - I am sceptical but open to some debate.

DragonFinlay, 2022-09-23:
I agree with most of what you say, but I have to admit that I don't understand everything since English is not my first language. I'm a mechanical engineer myself, I learned all this stuff in school a while ago, but I'm definitely not fully confident in my knowledge anymore without getting back into these topics. But thanks for your explanation, I found it interesting and it makes sense.

One argument I want to highlight is that, as you say, we're talking about 3D printed, orthotropic parts, not homogenous, isotropic parts. I too lack the knowledge to be confident about the theoretical part of all of this, but the practical part (I have printed this damn screw way too many times) showed me that the infilled print was noticeably stronger than the solid print, both for axial and torsional stress.

About the brittleness, I again refer to practical tests. In my case small, flat PETG parts with 100% infill vs. 20% infill. The solid parts sometimes splintered off the steel sheet when bending it whereas the hollow ones were very difficult to break (as they should be). But of course, there are many more factors in play here as you already mentioned.

Back to the original issue, did you try moving the text further up to only touch the bottom object, not intersect it? The remaining voids should not interfere with the visual quality at all, they are way too far inside the perimeters for that.

alidaf, 2022-09-24:
@DragonFinlay Anyone that has done Mech Eng has my respect. I still believe that it is one of the hardest courses anyone can do. I also respect you taking on the discussion as a second language and it could explain some of my initial contention with choices of word.

I haven’t tried any workarounds since. I strongly believe that 100% infill shouldn’t cause any artefacts but I do understand the intent to create contiguous volumes of infill. The piece I printed was a commission where I would normally use a more rounded font and haven’t had any previous issues. In this case I had to use a graphic that has sharp corners where the volume in those parts obviously falls foul of the ratio. I worked around that one by adding the image as a part (essentially at the same height as the base so no intersection as you suggested) but it raised the question as to why that was different to using the whole object, hence why I posted this report.

I appreciate your discussions thus far and hope this progresses to a developer taking this issue on board, even if it is to pacify my OCD for things to be correct!

DragonFinlay, 2022-09-30:
Thanks, I really enjoyed the courses and I like to think back to it. Even though I don't remember everything, it's extremely valuable to have the core knowledge anchored in your head, I'm sure you know what I mean.

I fully understand your point (and your OCD) - maybe we can have some automatic boolean operations in the future which properly combine parts without the need to carefully place it on top. Thanks for reporting!

closed 2026-09-29 by prusaslicer-bot[bot]