prusa3d/PrusaSlicer #8819: Height Range modifier incorectly displays height in Viewport

devilzmods, 2022-09-07:
### Description of the bug

In 2.5.0-rc2 the height range modifier displays the entered height incorrectly. 

It seems to round down to the next full integer, which had me very confused putting a modifier range from 0 to 1.8mm on a 2mm part. This seems to be a purely visual glitch, the part slices correctly with ironing turned on until just before the topmost layer.

[image]
[image]

Because that's a part that I don't want to share, the attached project just uses a primitive shape, the problem looks to require no extra steps to replicate. The issue affects both the bottom and top plane of the modifier range

[image]

### Project file & How to reproduce

[height_range_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/9505914/height_range_bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc2

### Operating system

Windows 10 Pro 21H2

### Printer model

Prusa Mini

DragonFinlay, 2022-09-08:
I think I found the issue, it only happens if the language is set to German
I guess that it comes from the fact that the German language uses comma "," as decimal separator, not period ".".

[image]

So the way these "planes" which visualize the height range are programmed probably doesn't account for the different separator.

bubnikv, 2022-09-08:
@YuSanka would you please check how the decimal separator is handled there?

YuSanka, 2022-09-08:
Thanks for your report. Fix for this bug will be a part of next (2.6.0) release.

FidelCapo, 2022-09-13:
As @YuSanka said, this issue should be fixed.
Closing.

closed 2022-09-13 by FidelCapo