prusa3d/PrusaSlicer #8822: Add "Coast" to Prusa Slicer

KomptonWest, 2022-09-07:
**What is Coast?**
Coast is a slicer setting that you can enable and configure to force the extruder to stop extruding material towards the end of its extrusion path. While the extruder is no longer pushing filament through the nozzle the build up of pressure and subsequent oozing of material ensures that the layer is completed. Coasting is an effective and **FAST** way to tune out artifacts like blips and blobs that manifest itself at the end of an extrusion and/or Z-Seams.

S3D Coast Video: [https://www.youtube.com/watch?v=WWpdGY0V-gM](https://www.youtube.com/watch?v=WWpdGY0V-gM)
Cura Coast Video: [https://www.youtube.com/watch?v=nLJf343xnqQ](https://www.youtube.com/watch?v=nLJf343xnqQ)

**What comparable "alternate" feature does PrusaSlicer have?**
PrusaSlicer recommended "Linear Advance" calibration which has the same net effect as 'Coast'

Prusa Slicer Linear Advance: [https://help.prusa3d.com/article/linear-advance_2252](https://help.prusa3d.com/article/linear-advance_2252)
Linear Advance Video: [https://www.youtube.com/watch?v=n3yK0lJ8TWM](https://www.youtube.com/watch?v=n3yK0lJ8TWM)

**A couple of observations on Coast vs Linear Advance:**

-Coast takes seconds and a few clicks to adjust.
-Coast is an easier concept for beginners to understand:
	-Example: See blips only at the end of an extrusion line?
		-Increase coast in the slicer GUI!

-Linear advance is an advance feature and a difficult concept for beginner and intermediate users.
	-Example: See blips only at the end of an extrusion line?
		-Print 'Before' test cube
		-Enable linear advance in Marlin
		-Print Calibration Patterns(s)
		-Save K value to Firmware and/or Prusa Slicer Start G-Code
		-Print 'After' test cube
		
**Special Note:** I would argue that Coast is needed in addition to Linear Advance and not in place of. Furthermore I would argue that Prusa Slicer does not have an alternative to Coast as Linear Advance is more of a firmware calibration feature.

**Has the community requested Coast as a feature?**
Yes! Multiple times as far back as April, 2017.

**Feature Requests from oldest to newest:**
-[Feature Request] Coast at end #246 Open...Apr 10, 2017
	-LINK: https://github.com/prusa3d/PrusaSlicer/issues/246
-Feature request: Please implement Coasting #583 Closed...Nov 21, 2017
	-LINK: https://github.com/prusa3d/PrusaSlicer/issues/583
-Adopt Simplify 3D Coast and Wipe capabilities #4526 Open...Jul 14, 2020
	-LINK: https://github.com/prusa3d/PrusaSlicer/issues/4526
	
**Has the community expressed an interest or desire for Coast via other channels?**
Yes! There are endless examples in various languages so I've listed just a few from English forums below.

**Prusa Forum Links**
	- [https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/how-to-change-the-coasting-setting/#post-184363](https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/how-to-change-the-coasting-setting/#post-184363)
	- [https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/how-to-change-the-coasting-setting/#post-184357](https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/how-to-change-the-coasting-setting/#post-184357)
	- [https://forum.prusa3d.com/forum/prusaslicer/coasting-in-the-gcode-no-coasting-setting-in-the-slicer/#post-606900](https://forum.prusa3d.com/forum/prusaslicer/coasting-in-the-gcode-no-coasting-setting-in-the-slicer/#post-606900)
	- [https://forum.prusa3d.com/forum/general-software-discussion/is-there-a-coasting-feature-in-prusaslicer-or-can-there-be/#post-561565](https://forum.prusa3d.com/forum/general-software-discussion/is-there-a-coasting-feature-in-prusaslicer-or-can-there-be/#post-561565)
	
**Reddit Links**
	- [https://www.reddit.com/r/prusa3d/comments/igkpep/coast_at_end_setting_in_prusaslicer/](https://www.reddit.com/r/prusa3d/comments/igkpep/coast_at_end_setting_in_prusaslicer/)
	- [https://www.reddit.com/r/prusa3d/comments/c4zgkf/where_is_the_coasting_setting_slic3r_pe/](https://www.reddit.com/r/prusa3d/comments/c4zgkf/where_is_the_coasting_setting_slic3r_pe/)
	- [https://www.reddit.com/r/prusa3d/comments/l414j9/how_to_set_retraction_coasting_and_wipe_distance/](https://www.reddit.com/r/prusa3d/comments/l414j9/how_to_set_retraction_coasting_and_wipe_distance/)
	- [https://www.reddit.com/r/3Dprinting/comments/uxf676/coasting_in_prusaslicer/](https://www.reddit.com/r/3Dprinting/comments/uxf676/coasting_in_prusaslicer/)
	
	
**Final Comments:** Anyone that has used coast in other slicers can attest to how useful that feature can be. Coast has the same net effect as Linear Advance but the time and effort to perform both varies greatly in favor of coasting. IMHO coast is as useful as retraction settings and as such I've created this post in hopes that this feature request gets some attention. Lastly, I'm aware that this feature request will likely get closed in favor of #246 but seeing that it is over 5 years old I was compelled to make a new, more compelling feature requests. Thanks!

gudvinr, 2022-09-08:
> -Coast takes seconds and a few clicks to adjust.

No, you still need to tune how aggressive coasting is per filament (same for LA) because if if would be one value fits all then LA wouldn't even exist in first place.

Main issue with coasting is that it is not a feature but a hack.  
You can't gradually decrease extrusion using only gcode and what it does is just halts extrusion before seam in hopes of oozing correct amount of filament.  

LA/PA implemented on firmware level and printer has more information about current speeds, acceleration, etc so it will provide better results. Slicer has very limited data and won't be as accurate as real hardware unless you add some printer simulator that uses actual firmware.

What you can do instead is demand from your printer manufacturer to enable LA in their firmware by default which is rarely the case.

KomptonWest, 2022-09-08:
@gudvinr - I didn’t realize that coast was such a polarizing subject. Why can’t we have both LA and Coast?

k9rdu, 2022-09-14:
For those of us with printers where LA is turned off and the source (or even the config.h) is not available (I'm looking at you, Creality!!! Ender 7 and CR-10 Smart here), as well as less advanced users that don't want to install an IDE, plug-ins, and compile and load their own firmware (my kid just wants a decent pikachu model and grandma wants a decorative hook for her apron)... this would be a "God-send"! ... hack or not. I've just spent 3 days and 2 rolls of filament (and counting) walking the fine line tuning between bulging corners and ringing. I'm a scientific/geeky guy that enjoys experimenting, but I am SO burned out on this.

gudvinr, 2022-09-14:
So... instead of asking for Creality to support LA you asking different company, that has nothing to do with Creality printers, and does have LA on theirs, to support feature that is useful for everyone but their users.  
Let that sink in.

KomptonWest, 2022-09-14:
@gudvinr - We’re trying to make a serious case for adding coast in addition to LA while you’re strictly arguing against adding coast because LA exists. There’s a big difference between the two positions that I can only hope you take the time to understand. Adding coast wouldn’t remove LA so the impact to you as an LA user is nothing. Furthermore, LA is a firmware feature not a slicer feature and I suggest YOU let that sink in first before you berate the end-users of the slicer for making use of the literal method that the slicer provides us users for suggesting the addition of new features.

gudvinr, 2022-09-14:
No, I don't argue against adding this and never said it is a bad idea. Your reasoning behind this idea is.  

PrusaSlicer has Prusa in it for a reason, same with Cura which is Ultimaker Cura. So, meaning of that is they should not care about other manufacturer's shenanigans if that does not benefit their business. Which is not unethical or something, it's just people who buy printers from Prusa pay for slicer development and others don't.

That being said, if you ever opened filament profiles for Prusa printers, you'll see they have K values for LA set in starting G-Code, which makes your argument for simplicity destroyed unless you have printer other than Prusa's.

So basically what you asking for is to spend time of developer which could be spent to benefit users who paid money on feature that benefits only people who bought printers from vendor that doesn't care even about toggling a switch in their FW.  
And these people for some reason reluctant to asking support from their vendor and ask for workaround from 3rd party company.

To be clear, I have no connections to Prusa so that's just my personal opinion but if you still do not understand what's wrong with your arguments then well, there's nothing anyone can do.

k9rdu, 2022-09-14:
You do understand that most first time 3D printer buyers don't drop the big
bucks for a prusa? ( I never even heard of them until I had been printing
for about 4 months)
... and more importantly to the case you claim to be making is that Prusa
did not make the software.

KomptonWest, 2022-09-14:
I have Prusa printers. I’ve used both LA and Coast extensively. I see value in having both (as do others - see entire post above). LA/K value is a predictive calculation based on ideal conditions which can drift for many reasons that require calibration and re-calibration. I’m not asking developers anything directly, I’m only making a suggestion on a forum that the developer made available for this exact purpose - making suggestions. As for the comments made @k9rdu He’s also within his right to make a suggestion for adding a feature for a printer that is explicitly featured inside Prusa Slicer.

VaTTeRGeR, 2022-11-12:
@gudvinr There are many people who use Prusa and Non-Prusa printers in the same lab. Not having to use separate slicers for your different types of printers would be a net positive thing for Prusaslicer, since it keeps users in the Prusa-ecosystem, thus making them more likely to buy a Prusa-made Printer in the future as well, as they will get attached to the slicer software.

Tupson444, 2022-12-23:
> So... instead of asking Creality to support LA you asking different company, that has nothing to do with Creality printers, and does have LA on theirs, to support feature that is useful for everyone but their users. Let that sink in.

Creality printers (and probably some other as well) for no good reason don't have Linear Advance enabled in firmware. Ok, so we can say that is an unnecessary flaw - and it is (it could be enabled and set to 0, so users can easily just adjust the value if they want to use it).

However, some people do use such printers that are what they are. In that case, coasting would be useful - not because it is a better option overall, but simply a useful substitute in the given circumstances (which are not rare).

This-guy-breaks-shit, 2023-05-10:
Ender 5 user here. My understanding is that the issue of linear advance in Creality printers stems from the particular stepper drivers used in the printers. Don't quote me on that, but I read into it a while ago. While it'd be lovely to upgrade my board and firmware for one that has the proper features, I can barely afford even the filament I use, so that's out of the question.

My take is that any and all features to improve accessibility are welcome. I'm no dev but I don't believe coasting would be enormously difficult to implement. It would make a welcome addition alongside linear advance for those who can use LA, and would be a welcome alternative for those who cannot. My experience with the Prusa i3 MK3S is that seams tend to bulge a little which can be detrimental to low tolerance gaps between moving parts within print-in-place mechanisms. The option to reduce or eliminate these artefacts would be wonderful.

lukasradek, 2023-07-10:
This would be greatly appretiated if it was implemented! 👍👍👍

cdburgess, 2023-07-27:
Until they add this feature (or not), I was able to find this solution that accomplished the same thing. https://www.reddit.com/r/3Dprinting/comments/uxf676/coasting_in_prusaslicer/

Harvie, 2023-11-20:
> Until they add this feature (or not), I was able to find this solution that accomplished the same thing. https://www.reddit.com/r/3Dprinting/comments/uxf676/coasting_in_prusaslicer/

Perhaps some of prusa slicer developers can confirm or disprove the solution mentioned on that reddit page to be equal to actual coasting?

This-guy-breaks-shit, 2023-11-21:
> Until they add this feature (or not), I was able to find this solution that accomplished the same thing. https://www.reddit.com/r/3Dprinting/comments/uxf676/coasting_in_prusaslicer/

I think I tried this. I didn't find it particularly effective, if at all. That said, since I began following this issue I have discovered that linear advance now works on creality 4.2.2 and 4.2.7 control boards (TMC2208 drivers) if you update to the latest version of marlin. There were changes made to the linear advance algorithm that have resolved the issues it causes with the aforementioned control board stepper drivers.

Coast would still be appreciated as an option mind you but for those with compatible creality printers, look into updating marlin to the latest version and enabling linear advance. You won't regret it.

lukasradek, 2023-11-21:
The problem is that linear advance is sometimes too advanced for a challenging setup. I have almost a meter long bowden setup, which requires ridiculous K value, which translates into "crazy" extruder motion, that is counterproductive. Coasting, while simpler, results in simpler extruder moves and sufficient result.

This-guy-breaks-shit, 2023-11-21:
Of course, I'm not trying to undermine the importance of coasting as a feature here. I'm just trying to help those in the position I was in when I found this thread: An ender 5 with a direct drive mod that still struggled with bloated corners and nonexistent consistency at seams. 

I understand that not all setups are compatible with linear advance or may need both features to print at an acceptable level of quality. Coasting is still a necessary feature option that prusaslicer needs.

Harvie, 2023-11-21:
> I have discovered that linear advance now works on creality 4.2.2 and 4.2.7 control boards (TMC2208 drivers) if you update to the latest version of marlin.

really? i was not able to find any such statement... do you know which commit? i wanna check if it's already merged in professional firmware ( https://github.com/mriscoc/Ender3V2S1/ )

This-guy-breaks-shit, 2023-11-21:
Apologies for not clarifying my source. I also do not know which version of Marlin began fixed the LA algorithm to work with TMC2208 drivers.

That said, I do believe there should be increased awareness about this for those on printers with 2208s. It's been a huge godsend for me in the absence of a coast feature in prusaslicer.

Harvie, 2026-09-02:
still relevant