prusa3d/PrusaSlicer #8823: Slicer adds a solid infill layer to a hollow area when importing STEP files.

Tanarin82, 2022-09-07:
### Description of the bug

When going to import a STEP file, the file appears to import properly, but when you go to slice it, the slicer adds a solid infill layer to a layer that should be hollow.

 
[image]

Left object, expected result (Object imported from an STL file.) Right object: STEP file version imported Object

### Project file & How to reproduce

[Body Tube Bottom STL vs STEP.zip](https://github.com/prusa3d/PrusaSlicer/files/9522140/Body.Tube.Bottom.STL.vs.STEP.zip)

Steps to reproduce: Open the .3mf file and slice

Alternatively, I have included the STEP file and STL file so that it can be tested with a fresh import of the files themselves. 

If you choose to import the files manually:
1: Import STEP file
2: Select Place on face and select the top face (This step is not needed as the issue still happens when you skip this step, but put here for completeness sake)
3: Slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10 21H1

### Printer model

Ender 3

bubnikv, 2022-09-08:
Sometimes the STEP import produces cracks in the model. If you are on
Windows, you may try to fix the model in slicer by the Netfabb service.

DragonFinlay, 2022-09-08:
It definitely seems to be related to the STEP file itself.
When exporting the STEP as STL in PrusaSlicer and checking for nonmanifold geometry in Maya, nothing is found.
However, reimporting that same exported STL also gives the same problem, so the STL format doesn't "get rid" of it.

bubnikv, 2022-09-08:
We are using an OpenCascade CAD kernel for STEP import. It is the same CAD kernel that is used by FreeCAD. I suppose FreeCAD would have the same issue importing the same STEP file. Unfortunately it is out of our realm to fix OpenCascade CAD kernel.

snafu1282, 2023-04-28:
I just ran into this issue with a case for AA batteries.

[image]

The STEP file is on the left, the one with the odd blue rings around only 4 of the holes. The right one is STL.

[image]

After slicing, the STEP version has the infill over the 4 holes, but the STL version doesn't.

Interestingly, this happened after I reduced the diameter of the holes from 15mm to 14.2mm. The 15mm version sliced properly.
The model was created in F360.

closed 2026-09-29 by prusaslicer-bot[bot]