prusa3d/PrusaSlicer #8825: ⚠️PrusaSlicer v2.5.0 Installer triggers security alerts

zigurana, 2022-09-08:
### Description of the bug

Upon installing PrusaSlicer v 2.5.0, I get an immediate call from my IT department, because the installer triggers a security alert.
This makes me and my employer rather unhappy.

The alert concerns POST requests to collect.installeranalytics[.]com (35.153.149.66):
[image]
[image]

### The raw alert message
`alert http $HOME_NET any -> $EXTERNAL_NET any (msg:"ETPRO MALWARE TakeMyFile Installer Checkin"; flow:established,to_server; content:"POST"; http_method; content:"AdvinstAnalytics/"; http_user_agent; depth:17; 
fast_pattern; content:"t="; http_client_body; content:"lc="; http_client_body; content:"v="; http_client_body; 
content:"aid="; http_client_body; content:"cid=";http_client_body; content:"sid="; http_client_body; http_header_names; 
content:!"Referer"; reference:md5,972c0a0eaf52b550b704fa2cc07a64bc; classtype:trojan-activity; sid:2849813; rev:1; 
metadata:affected_product Windows_XP_Vista_7_8_10_Server_32_64_Bit, attack_target Client_Endpoint, created_at 2021_09_01, deployment Perimeter, former_category ADWARE_PUP, signature_severity Minor, updated_at 2022_05_03;)`

### Installer details
Name: PrusaSlicer_Win_standalone_2.5.0.exe
Size: 86208296 bytes (82 MiB)
CRC32: 5F63E531
CRC64: B5DA5BD8398E48D9
SHA256: 1A417FE95D2EEBD7E0D5A4A2A51E043BA594A865F223841A45CA6D4E88E9444C
SHA1: AC509AFCFA2C9688AC5C4400928AC9564EACDFD8
BLAKE2sp: 4B12C23A65B2B2B244068C98E94E90D3F2195216FCD31E1F9AB1A14A71A7102B

### Reproducibility
This issue was reproduced with company IT on the line, and installing / uninstalling both triggered the callbacks.

### Project file & How to reproduce

Not applicable, this involves SW installation only. PrusaSlicer was not opened for this issue to occur.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Prusa Mini+

bubnikv, 2022-09-08:
We will check with the college who prepared the Windows installer for us. We are using a 3rd party commercial application to build our PrusaSlicer installers. We will check with the vendor of the installer.

Janys, 2022-09-08:
Hello, I'm the person responsible for the installer and I'm sorry for the trouble it caused - this is definitely not the expected behavior. I checked the project file in Advanced Installer (the tool we use to build installers) and found out that the last update of the application enabled a network service that was previously disabled.

I have fixed this issue and compiled a new installer with all network features strictly off. It is already uploaded on our server, so if you redownload the file, it should no longer "call home".

kubispe1, 2022-09-08:
Ok fine, thank you @Janys very much for cooperation, so I close it. 
@zigurana  in case there will be some trouble, let us know

closed 2022-09-08 by kubispe1

zigurana, 2022-09-08:
Thanks for picking this up so quickly!
I'm kind of afraid to test it on the company network again, but I'll take your word that the issue is fixed now.

I am left with the following questions though: when and what was communicated? And was it only active when running the installer? (so independent of PrusaSlicer itself?)

bubnikv, 2022-09-08:
A 3rd party installer creator created an installer which called home. We were not aware of that default behavior of the 3rd party installer creator, we found a switch to disable it.

Janys, 2022-09-08:
> Thanks for picking this up so quickly! I'm kind of afraid to test it on the company network again, but I'll take your word that the issue is fixed now.
>
> I am left with the following questions [...]

This is an official built-in feature of the [Advanced Installer](https://www.advancedinstaller.com/) that offers some very basic tracking (like the number of installs and OS version) for tech support purposes during the installation. However, this would require us to set up a paid plan with the Advanced Installer company - which we don't have because it doesn't make sense to us, so it was enabled purely by my mistake and oversight.

PrusaSlicer has an opt-in hardware survey which shows a window and asks you whether you would like to share your system data with us so we can improve the software further. This is much more valuable for us.

We also have an old-school Inno Setup installer, which is more basic and it only checks for updated packages via a simple HTTP request. It's also available at prusa3d.com/prusaslicer and it's called Drivers and Apps 2.5.0.

Fusion3Design, 2023-03-02:
Could you share the .aip file for your advanced installer project? We are trying to build a new version of the slicer and would like to avoid some of these pitfalls you have already identified. Thank you very much