prusa3d/PrusaSlicer #8826: none prusa printers have no filament listed

midicdj1000, 2022-09-08:
### Description of the bug

I have many printers setup and after upgrading to 2.5 only prusa printers show filaments in the drop down. I have told the setup that all printers should use all filaments, but they have * marked and it says

 Filaments marked with * are not compatible with some installed printers

Why is this ? 

[image]
[image]

So all of my creality printers have no filaments I can use including any that I had setup before.

### Project file & How to reproduce

choose a none prusa printer and the filaments section is empty

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

windows 10

### Printer model

creality cr20, ender 3, ender 6 cr10

midicdj1000, 2022-09-08:
If I choose a system preset printer the filaments show up, but user preset shows no filaments.
I have user profiles setup for revo nozzle size for example.

DragonFinlay, 2022-09-08:
I added the CR-20 Pro and tried to reproduce it, but I can't reproduce it, I get a nice list of compatible filaments.
Can you attach the 3MF with your printer preset please so we can check the keywords and dependencies? Maybe we can find out more from that.

DragonFinlay, 2022-09-08:
I'm not entirely sure how it was before updating PrusaSlicer, but the printer profile from your 3MF was detached from the original system profile and therefore the dependencies don't work anymore.

Setting your profiles up from zero is probably the cleanest solution.
Otherwise there would be a workaround showing the whole list of filament presets, no matter if they are deemed compatible or not. For this, check "Show incompatible print and filament presets" in the preferences and click the green flag next to the dropdown menu on the filament tab.

Since this is profile related and not an issue with PrusaSlicer per-se, I will close the issue.

closed 2022-09-08 by DragonFinlay

midicdj1000, 2022-09-08:
Show incompatible print and filament presets" in the preferences and click the green flag next to the dropdown menu on the filament tab.
this is showing my old filament profiles so is good enough for me, thank you for the help.

gearmesh, 2022-11-15:
I have the same issue. I have a Original Prusa i3 MK2S and now all the filament choices are blank, including the Prusa filaments.