prusa3d/PrusaSlicer #8827: Error Parsing Int on Pressure Equalizer

yasaslive, 2022-09-08:
### Description of the bug

Cannot set a value in pressure equalizer settings(both positive and negative values)
[image]

### Project file & How to reproduce

Open PrusaSlicer -> Go to Print Settings -> under pressure equalizer set values to 10 on positive and negative on volumetric slopes.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11

### Printer model

Tevo Tarantula (Klipper)

DragonFinlay, 2022-09-08:
It works fine with other models, so it's most likely related to the settings or the 3D model you're using.
We would please need the 3MF with the model in order to reproduce it.

hejllukas, 2022-09-13:
I wasn't able to reproduce it, probably because your attached 3MF isn't produced by PrusaSlicer. So there aren't print setting that PrusaSlicer is saving into 3MF. Probably this issue is happening only with some specific settings.
So please upload 3MF project file from PrusaSlicer **File->Save Project as...**, it should help us to reproduce it.

yasaslive, 2022-09-14:
Sure, [Here it is!](https://drive.google.com/file/d/10e2nscrdwecdDHk_L-WdXkNX9a46sKm2/view?usp=sharing), Also added [another one](https://drive.google.com/file/d/1sjJtXctTMaNach9smMS5omXB1Y-nQyxP/view?usp=sharing) just in case

hejllukas, 2022-09-15:
Thank you! I was able to reproduce it. The pressure equalizer has the issue with the tag TIMELAPSE_TAKE_FRAME in the "Before layer change G-Code". It will be fixed in next release.

locki-cz, 2022-10-13:
great! I hope for some beta soon! :)

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena

voland696, 2023-02-08:
2.6.0-alpha3 is still crushing if TIMELAPSE_TAKE_FRAME in the Before layer change G-Code (access violation error)

lukasmatena, 2023-02-08:
@voland696 Thanks. There are actually two places where the line was causing a crash. One was the pressure equalizer which is discussed here and was fixed by @hejllukas. The other one was cooling buffer, which was reported in #9550 and which was fixed by me, but only after 2.6.0-alpha3.