prusa3d/PrusaSlicer #8829: Too many wipes and retraction under certain conditions

Asariuss, 2022-09-08:
### Description of the bug

PrusaSlicer can perform even 5 retractions one after another before starting at new point creating under extrusion.

### Project file & How to reproduce

[image]
Open attached 3mf file in zip archive and navigate to gcode line presented in screenshot
[enclosure.zip](https://github.com/prusa3d/PrusaSlicer/files/9530193/enclosure.zip)
Effect:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Creality CR-6 SE

beeb, 2022-09-09:
Possibly related, but I have inconsistent wipe behavior when printing a very simple 2-perimeters wide object, see below.

Expected behavior (no wipe between outer and inner perimeter):
[image]

Next layer (wipe between the two perimeters):
[image]

beeb, 2022-09-09:
@Asariuss after looking at your code I don't think you have the bug I mentioned. It seems there is a problem with too many wipes, but the Gcode you are showing is just one wipe sequence split into 5 moves.

Asariuss, 2022-09-09:
If its intended behaviour, then i should look into retract before wipe. I think I misunderstood what wipe is. Thanks!

> @Asariuss after looking at your code I don't think you have the bug I mentioned. It seems there is a problem with too many wipes (see my screenshots), but the Gcode you are showing is just one wipe sequence split into 5 moves. You probably just have a problem that the wipe is too long or retraction too hard.

DragonFinlay, 2022-09-17:
@Asariuss What beeb says is correct, it's a single wipe but since there's not much space and it should follow the infill line, it's split into 5 segments / code lines.

The issue on your photo doesn't look like it could come from anything inside your 3 perimeters - it rather looks like the result of seams being aligned there and printing external perimeters first, possibly with a few other print settings interfering as well. If there's nothing forcing you to print external perimeters first I'd recommend disabling that setting.

@beeb Could you please share the 3MF file as well? The G-code doesn't give me the 3D model to reproduce it.

apalert, 2023-05-16:
The distance between the seams in first pic is much less then in the second. Perhaps it was less than the Minimal Travel setting therefore didn't trigger the 2nd retract/wipe sequence. I usually set Min Travel to 0.1mm in this situation.

[image]

Asariuss, 2023-05-18:
> @Asariuss What beeb says is correct, it's a single wipe but since there's not much space and it should follow the infill line, it's split into 5 segments / code lines.
>
> The issue on your photo [...]

For sure im not using external perimeters first but i have aligned seams. Its problem of nozzle getting empty some situations. Mostly when printing some boxes (walls etc.). I reduced "retraction before wipe" to 0%-5% and retraction itself which helped but not completely. 
Problem is that not every model have it. Maybe its extrusion problem? Im slowly investigating it. 
Had this problem again with this model:
https://cults3d.com/en/3d-model/tool/airsoft-unloader
I have no photos of outcome but I think in this example problem occurs because gapfilling at corners. If I have time to isolate and test this i will post my results for future readers.

beeb, 2023-05-22:
> Perhaps it was less than the Minimal Travel setting

@apalert good observation, but then the question becomes, why is the seam not on the same position for two consecutive layers with very similar geometry? Also, in my case I would rather not have the retraction and thus maybe I would increase the minimum travel after retraction to something larger.

closed 2026-09-29 by prusaslicer-bot[bot]