prusa3d/PrusaSlicer #8830: slicer misses holes

wbrokow1, 2022-09-08:
### Description of the bug

PS missed holes in z direction when sliced
[x_carriage_frame_left (1).zip](https://github.com/prusa3d/PrusaSlicer/files/9530212/x_carriage_frame_left.1.zip)

### Project file & How to reproduce

load file to plater 
pick settings etc. 
slice

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 10 64 bit

### Printer model

delta kossel smoothieboard

neophyl, 2022-09-08:
Erm you have slicing mode set to Close Holes.  Print Settings>Advanced>Slicing Mode.  

What did you expect a mode called Close Holes to do ? lol

wbrokow1, 2022-09-08:
That was the default . I didn’t set it.lol

Walter Brokowski

neophyl, 2022-09-09:
I don't know where you got your profiles, but as a quick check I installed a Anycubic Kossel Linear printer using the wizard in 2.5 and all the default print profiles added for that have Regular as the slicing type default.  I couldn't find your specific printer in the wizard.  
That setting hasn't been there that long but everything I've seen all has it set to Regular.

foreachthing, 2022-09-09:
I did not ask @BoevoyNarod, but I cannot imagine that he/she changed that setting on purpose - in https://github.com/prusa3d/PrusaSlicer/issues/8820 ... so, I'd believe Walter that he has never changed that setting.

neophyl, 2022-09-09:
So there might be an issue with some of the profiles changing values in the new software then ?

Must admit its not one I have ever used either.  I spotted it as I changed the print profile to my normal one for testing and it sliced fine, then I did a compare using the profile compare tool and it sort of was obvious at that point.

rtyr, 2022-09-09:
Did you use your own default settings or settings included in some downloaded 3MF project?

BoevoyNarod, 2022-09-18:
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/9593247/PrusaSlicer.zip)
Unfortunately as in my #8820 I catch the same again and this time Slicing mode is "regular" but the hole has been lost

rtyr, 2022-09-19:
It can also happen with broken model, you can try to repair it with netfabb service.

DragonFinlay, 2022-09-27:
Closing as completed

closed 2022-09-27 by DragonFinlay