prusa3d/PrusaSlicer #8831: COMPENSACION XY DAÑADA

djinnuy, 2022-09-08:
### Description of the bug

BUENAS TARDES, INSTALE LA ULTIMA VERSION 2.5, Y TIENE UN BUG, COMPENSA PARCIALMENTE LOS PERIMETROS, CUANDO LE DAS COMPENSACION XY TE COMPENSA UNA PARTE DE LOS PERIMETROS Y LA OTRA NO LA TOCA, EN LA VERSION ANTERIOR FUNCIONABA IMPECABLE, LES ENVIO CAPTURA DE PANTALLA  
[image]

### Project file & How to reproduce

ABRAN UN CORTANTE PARA GALLETAS E INTENTEN COMPENSAR SU PERIMETRO PARA VER LA FALLA
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 WIN64

### Operating system

WINDOWS 10

### Printer model

ARTILLERY X1

neophyl, 2022-09-08:
The new slicing engine by default is Arachne, change it to Classic to get the previous behavior.  Print Settings>Layers and Perimeters>Advanced>Perimeter Generator

djinnuy, 2022-09-08:
Thank you alot

Obtener Outlook para Android<https://aka.ms/AAb9ysg>