prusa3d/PrusaSlicer #8835: The icon on the taskbar is blank

Mng-Michael, 2022-09-09:
### Description of the bug

The icon on the taskbar is blank

downloaded here: [https://www.prusa3d.com/page/prusaslicer_424/]
[image]

### Project file & How to reproduce

open the app

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa MK3S

ajdavids, 2022-09-10:
I have the same issue! Didn't with 2.4.x or with the 2.5 betas, but I do now with the full release.

gaminnick, 2022-09-11:
I had this issue with 2.4 and 2.5.  I have PrusaSlicer pinned to my taskbar and when I run it, a new, blank icon is created on the taskbar for that running session.

[image]

rubegartor, 2022-09-12:
I have had the same problem after upgrading from 2.4 to 2.5 in Windows 11, however I have managed to set the icon so that it does not appear blank.

1. Click properties:
[image]

2. Browse icons of prusa-slicer-console.exe located at C:\Program Files\Prusa3D\PrusaSlicer in my case, and apply and accept:
[image]

I hope it works for you

gaminnick, 2022-09-12:
Awesome!! That worked.  And to top it off, I pinned the running application icon to my taskbar, removed the original one I had made and now it does not create an extra icon.  And if I right click and open another session it now stacks as it should.  I did have to rename it as it put a (2) at the end of the name since I already had an icon on the taskbar.  You could probably remove it and run it from the Start->Apps and then pin that one.

[image]

DragonFinlay, 2022-09-20:
Issue seems to be related to Windows rather than PrusaSlicer. Closing.

gaminnick, 2022-10-02:
I think I found the issue.  The problem seems to be how you are starting PrusaSlicer.  The icon you generate is really starting prusaslicer-updater.exe.

"C:\Program Files\Prusa3D\PrusaSlicer\prusaslicer-updater.exe" /silent -restartapp prusa-slicer.exe -startappfirst

I think this causes PrusaSlicer to start separately with the blank icon and puts an additional icon on the taskbar when running.  So this is not a Windows issue.

Neophyte69, 2023-06-26:
I have now experienced this issue with Windows 11, but I was able to resolve it without major hassle.

When opening PrusaSlicer from the "Programs" list, it shows a blank icon on the taskbar.

However, when opening it from the desktop shortcut, it does show the correct icon on the taskbar.

jdon759, 2023-09-27:
> When opening PrusaSlicer from the "Programs" list, it shows a blank icon on the taskbar.
> 
> However, when opening it from the desktop shortcut, it does show the correct icon on the taskbar.

I have the exact opposite issue with PrusaSlicer 2.6.1 on Windows 11 - The icon shows correctly when launching from start menu, but does not show when launching from desktop shortcut.