prusa3d/PrusaSlicer #8836: Windows Installer ignores "Create a desktop shortcut" checkbox

beeb, 2022-09-09:
### Description of the bug

When installing the latest 2.5.0 release of PrusaSlicer on Windows (10), the installer asks if the user wants a desktop shortcut created. Disabling the checkbox still results in 2 desktop shortcuts being created (for PrusaSlicer and G-code viewer).

### Project file & How to reproduce

Install PrusaSlicer on Windows and uncheck the desktop shortcut box. Observe a shortcut has been created after install.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10 Pro

### Printer model

N/A

rtyr, 2022-09-09:
@Janys FYI

lukasmatena, 2024-02-19:
Should be solved. Closing.

closed 2024-02-19 by lukasmatena