prusa3d/PrusaSlicer #8837: Crash when slicing on Mac

capi92, 2022-09-09:
### Description of the bug

Systematic crash when slicing the attached project.

Version: 2.5.0+MacOS-arm64
Build: PrusaSlicer-2.5.0+MacOS-arm64-202209060713
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 12.4 (Build 21F79)
Total RAM size [MB]: 17,180MB

### Project file & How to reproduce

Slice the attached project file.

[Laser screen.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9534378/Laser.screen.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+MacOS-arm64-202209060713

### Operating system

macOS 12.4 (21F79)

### Printer model

Prusa MK3s

FidelCapo, 2022-09-09:
Thank you for report. We can reproduce crash on MacOS and Win.

capi92, 2022-09-09:
Excellent, do you know how to avoid it waiting for the fix ?

3DCoded, 2022-09-09:
It slices fine for me on 2.5.0-rc2 on macOS 12.5.1 with M1 Chip and 16GB RAM.

capi92, 2022-09-09:
> It slices fine for me on 2.5.0-rc2 on macOS 12.5.1 with M1 Chip and 16GB RAM.

Bug report is not on rc2 but final release

FidelCapo, 2022-09-12:
Not at the moment, unfortunately. We're figuring out what's wrong.

FidelCapo, 2022-09-16:
This issue should be fixed in the next release.
Closing.

closed 2022-09-16 by FidelCapo