prusa3d/PrusaSlicer #8839: Ironing feature is ironing layers that is not top layer

rahmanshaber, 2022-09-09:
### Description of the bug

It's hard to explain. I selected the ironing in the top surface only. It also ironed the whole layer even where there will be another layer. as a result there was not much fusion happend between the ironned layer and the next, so the upper part of the body breaks of easily

See here how the layer is not rough, so the next later was not attached to it properly.
[image]

See those 2 hexagon feature also broke of and the surface of the broken side is smooth.
[image]

This is how the part is.
[image]

### Project file & How to reproduce

all the settings were default except the ironing is selected to "All top surfaces"

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.5

### Operating system

windows 11

### Printer model

MK3

rtyr, 2022-09-16:
[v5.1-Body_top001.zip](https://github.com/prusa3d/PrusaSlicer/files/9583070/v5.1-Body_top001.zip)

rtyr, 2022-09-16:
Likely similar ssue as https://github.com/prusa3d/PrusaSlicer/issues/5634, https://github.com/prusa3d/PrusaSlicer/issues/5000. Can be caused by heatcreep or insufficient pressure in the nozzle after ironing.

rahmanshaber, 2022-09-16:
@rtyr  I don't know what's heat creep. Please look at the pictures, you can see it's ironing the whole layer even where it's not the top layer.
[image]

That marked area is not top layer. but it ironed that area and maybe it makes harder to fuse with the next layer. Or ironing that layer reduce the Z height of the part and next layer is poured slightly above the actual layer.

rtyr, 2023-09-05:
It is top layer.

rtyr, 2023-09-05:
I believe I understand now, but I don't see any ironing in the preview. Maybe your model had errors and there was a missing layer.

Anyway can you please try with the latest slicer?

rahmanshaber, 2023-09-05:
glad to hear that.
in the picture you can clearly see the ironing part. 
and unfortunately i don't have access to printer now. i will inform you when i print it again. 

> Maybe your model had errors and there was a missing layer.

You can see my model file, i don't think any layers are missing.

rtyr, 2023-11-23:
Not reproducible with current slicer version. Closing.

closed 2023-11-23 by rtyr