prusa3d/PrusaSlicer #8841: Rear Seam Not Aligned, Not Rear of Model

Aero-Grip, 2022-09-10:
### Description of the bug

Same .3mf file in 2.4.2 has the seam position set to rear, slicing lines it up at the rear-most Y location for that layer (mostly). 2.5.0 has the seam wandering up the curved surface and not at the rear most Y for that layer.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10 Pro 

### Printer model

Prusa Mk3s+

Aero-Grip, 2022-09-10:
I should note changing from Arachne to Classic had no effect on the seam wandering.

zmeygeruch, 2022-09-10:
I have this bug roo.

zmeygeruch, 2022-09-10:
I have this bug too.
Set the seam to the "rear side"
How it looks on 2.4.2 prusa
[image]
and on 2.5.0
[image]

Witawat, 2022-09-12:
me too
[image]

Garryb123, 2023-01-11:
> I have this bug too. Set the seam to the "rear side" How it looks on 2.4.2 prusa [image] and on 2.5.0 [image]
> 
> Attached file with img and .3mf [3mf_zip.zip](https://github.com/prusa3d/PrusaSlicer/files/9541318/3mf_zip.zip)
> 
> P.S. Tested on Win10

I have this bug too and it really is a problem for some parts. SuperSlicer now has the same bug and adding a seam modifier has stopped placing the seam where intended. Obviously I want this fixed as soon as possible; it seems it should be easy.

zmeygeruch, 2023-01-11:
> 

Just disable Arachne engine and all be ok for now :)

Garryb123, 2023-01-11:
Thank you. I'd tried that but the seam still is placed in a corner.

[image]

[seamtest.zip](https://github.com/prusa3d/PrusaSlicer/files/10395828/seamtest.zip)

zmeygeruch, 2023-01-12:
> Thank you. I'd tried that but the seam still is placed in a corner.
> 
> [image] [image]
> 
> [seamtest.zip](https://github.com/prusa3d/PrusaSlicer/files/10395828/seamtest.zip)

Hi. You can download the older version 2.4.2.
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2
It's work for me.

Garryb123, 2023-01-12:
> Hi. You can download the older version 2.4.2. https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2 It's work for me.

Thank you. Since Arachne isn’t important to me I'll keep using SuperSlicer 2.4.5 because the bug isn’t it it. But this seems like such an obvious regression in PrusaSlicer I can't understand why no one has even been assigned yet to fix it.

SachCZ, 2024-10-25:
Hello, this issue is fixed and now the behavior is very similar to PrusaSlicer 2.4.2.

closed 2024-10-25 by SachCZ